FastLED/FastLED #459: WS2811S chipset Support

Nik1990, 2017-05-31:
I've a strip led with WS2811S (note the "S"). Is the chipset supported from the library?
With the following code (blink.ino) 

`#include "FastLED.h"

// How many leds in your strip?
#define NUM_LEDS 60

// For led chips like Neopixels, which have a data line, ground, and power, you just
// need to define DATA_PIN.  For led chipsets that are SPI based (four wires - data, clock,
// ground, and power), like the LPD8806 define both DATA_PIN and CLOCK_PIN
#define DATA_PIN 13

// Define the array of leds
CRGB leds[NUM_LEDS];

void setup() { 
      FastLED.addLeds<WS2811, DATA_PIN, RGB>(leds, NUM_LEDS);
}

void loop() { 
  // Turn the LED on, then pause
  leds[0] = CRGB::Red;
  FastLED.show();
  delay(500);
  // Now turn the LED off, then pause
  leds[0] = CRGB::Black;
  FastLED.show();
  delay(500);
}
`
the strip led now work (random colors)

Thanks

focalintent, 2017-05-31:
I haven’t done anything explicit to support it - and a quick search for a data sheet to get the timing isn’t turning anything up.  If you can find a data sheet for it (that isn’t the WS2811 data sheet) then I can add support for it.

Nik1990, 2017-05-31:
Hi, i have found only this

http://www.world-semi.com/solution/details-106-4.html

but after opened data-sheet it seems that ws2811 and ws2811s are the same.

focalintent, 2017-05-31:
Also what hardware are you testing this on?

jasoncoon, 2017-05-31:
The WS2811S seems to be the same chip (and protocol), just in a SOP-8 (small outline package), as opposed to DIP-8 (dual in-line package).

Nik1990, 2017-05-31:
With a Arduino Uno. Data pin connected to PIN 13.

focalintent, 2017-05-31:
What happens if you try a different pin?  (Pin 13 is attached to the onboard led - I wonder if that is messing with signal timing)

focalintent, 2017-05-31:
Also what version of arduino are you using?

Nik1990, 2017-06-01:
So, i connected the DATA Pin of strip LED to pin 8 of arduino and the problem is the same. Arduino version is the "Arduino UNO". But it is a not original version. It is possible that the problem is this?

marmilicious, 2017-06-01:
No, what version of the Arduino IDE.  1.8.0,  1.6.8,  etc.?

Nik1990, 2017-06-01:
This:

[image]

focalintent, 2017-06-01:
This is not a version of arduino that I have tested with - it is possible that they have made compiler changes that are interfering with the asm code for the WS2811.  Try going back to 1.6.x or 1.8

electrokean, 2017-06-01:
@focalintent @Nik1990 I believe that is Arduino 1.0.5 packaged for Debian Linux - pretty old, but Arduino don't allow third party distribution of newer versions so you have to download and install a newer version from arduino website, not via the Linux package manager

focalintent, 2017-06-01:
Ah I misread, looked like it was 2.0.5 - that said
Arduiono 1.0.x hasn’t been tested/supported in a long time.  Try getting 1.6.x or 1.8.x (the readme for the library even explicitly states this)

Nik1990, 2017-06-01:
I have just tried Arduino 1.6.0 under windows and the problem is the same. Same sketch only changed the pin (12 to 4).

MFornander, 2017-06-01:
Could you connect a normal WS2811 or WS2812 strand in the exact same place so we can see if it's hardware or software?

Nik1990, 2017-06-01:
Just tried also Arduino IDE 1.8.1. Same problem. And no, i don't have WS2811 or WS 2812. Only 5m of this chipset strip led.

electrokean, 2017-06-01:
@Nik1990 It is possible the WS2811S that you have are 400kHz version - can you try an LED type of WS2811_400
In other words, change your setup code to:
`FastLED.addLeds<WS2811_400, DATA_PIN, RGB>(leds, NUM_LEDS);`

As you don't have any other strips to test, you could instead do some testing using the Adafruit NeoPixel library and see if they work with that, either as NEO_KHZ800 or NEO_KHZ400

Nik1990, 2017-06-01:
@electrokean Nothing, i tried WS2811_400 and Adafruit NeoPixel library. Same problem.

electrokean, 2017-06-01:
Hmmm... Maybe your LED strip is faulty - or at least the first chip/LED, which I've seen happen before.
Can you desolder or cut the strip into 1m sections and test the separately?

focalintent, 2017-06-01:
How do you have things wired up?  This is beginning to sound like either a problem with you led hardware or your wiring.

Nik1990, 2017-06-01:
[image]

For debug i cutted only 3 led (1 chip) connected to 12v 3A power supply. All library seems to show random colors with simple code. I also tried a 5v 2A power supply and only red led switch on. Tried also the @electrokean  suggest of test other first chipset of strip and the problem is the same.

Blink515, 2017-06-01:
You need to connect the ground from your power supply to the strip and the
Arduino board.

focalintent, 2017-06-01:
You need to also connect ground on the led strip to ground on the arduino - it's not enough to just connect the data line.

Nik1990, 2017-06-02:
Thanks boys :D. Two days for a wire, problem solved.

closed 2017-06-02 by Nik1990