FastLED/FastLED #460: FastLED Instability on ESP8266 (NodeMCU 1.0) + 12v WS2811 LEDs

bruhautomation, 2017-05-31:
Hello! 

Thanks much for your efforts on creating the FastLED library. It's awesome! I used it to create a MQTT LED strip controller for Home Assistant.  https://github.com/bruhautomation/ESP-MQTT-JSON-Digital-LEDs. When it is working, it is awesome!

Me and several other people are fighting a chronic instability with the ESP8266 (NodeMCU1.0 dev board) -> 3v-5v logic level shifter -> WS2811 LEDS. The code runs great for about 1-3 hours, but then throws a WDT reset and the chip restarts. After that, everything is responsive, but it kills the effect until it is called again. I have been very unable to diagnose the bug. 

Here are a few things I know. 

1) If I use the same code with 5V Adafruit Neopixel rings I don't have any problems. Stable for >24 hours. 

2) I run the code on NodeMCU 1.0 and am not writing the LED values to the strip via FastLED.show() the code is stable in the loop and does not crash. As soon as I enable FastLED.show(), the strip will be fine for an hour or two and then crash. 

My assumption is that there is some kind of odd interaction between the ESP8266 and the 12v WS2811 LEDs.  

Do you have any guess as to what could cause this hardware issue? I know that hardware SPI is not currently supported on the ESP8266 chip. Could that potentially resolve the issue? 

Also, any plans on supporting the ESP32? I know that's crazy work. Just wanted to see if you were interested. 

Thanks much for all your work and efforts on this project. It's awesome!

focalintent, 2017-05-31:
How many leds when 12v vs. how many leds when 5v?  Also - are you disabling interrupts?  WDT is the watchdog timer - if interrupts are disabled and you are writing a large number of leds, you increase the chance that something is going to end up unhappy while interrupts are disabled.

Hardware SPI won't help here - FastLED doesn't use SPI to talk to WS2811's.

As for the ESP32 - that's on the list to eventually support, but everything in the library is on hold right now - as it says on the front page:

*NOTE* library work is currently on hold - see https://plus.google.com/102282558639672545743/posts/JQKcujjaXXC for details.

bruhautomation, 2017-06-03:
Thanks for the response. I'm using ~180 LEDs. It seems removing these lines from my code greatly increased my stability. 
```
#define FASTLED_INTERRUPT_RETRY_COUNT 0 
#define FASTLED_ALLOW_INTERRUPTS 0
```
Sorry to hear about the loss of a loved one. Thanks much for all your work on this repo!

closed 2017-06-03 by bruhautomation