FastLED/FastLED #465: Change WS2812 timings to include the 300us WS2813 wait frame

MFornander, 2017-06-13:
It seems like new WS2812 strips may actually be WS2813:
https://blog.particle.io/2017/05/11/ws2812b-neopixels-are-about-to-change

To be safe should we change the timing of all WS2812 code to have a default 300us WAIT?  I'm expecting an ever increasing flood of "my LEDs are flickering" issues being opened.

focalintent, 2017-06-13:
This is a little tricky because right now that value is also used to work out the buffer for allowing interrupts, so it won’t _just_ be changing the number - it’ll break interrupt code to have the wrong value in there.  But it’ll be dealt with.

MFornander, 2017-06-14:
First of all thanks for all the awesome work here Daniel, especially juggling WS timings and interrupts.

I've been battling interrupts on the ESP8266 for months now.  Disallowing interupts locks my ESP sooner or later so luck there.  What's great is that with the 300us WAIT of the WS2813 setting, I'm getting a much more stable system which makes sense since it now has more time to survive the ESP's WiFi interrupt.

An idea here would be to allow/guide interrupts toward the high signal since it seems like there is no time limit on holding the signal high.  i.e. we could send a few bits, wait for a T1H and hold it there to allow interrupts.  What's the overhead for blocking and unblocking interrupts on the ESP you know?  Maybe we could nudge/force the ESP Wifi/Watchdog during the T1H moments?

Unlimited time on TH source:
https://wp.josh.com/2014/05/13/ws2812-neopixels-are-not-so-finicky-once-you-get-to-know-them/

MFornander, 2017-06-14:
Oh I just got what you were saying about interrupts I think after looking at the ESP code.  By making the default WAIT 300us when a WS2812 will only tolerate 50us, we will actually RESET by mistake if we get a long interrupt on an old 50us WS2812...

focalintent, 2017-06-14:
Unfortunately that "unlimited hold high" time is only valid for input to the first led - the long-held-high signal isn't carried through the re-shaped output that the LEDs have, so that isn't an option.

closed 2019-01-20 by focalintent