FastLED/FastLED #466: Display contine

jay1384, 2017-06-16:
Hi,
In Fast LED library we define number of LEDs as #define NUM_LEDS 100. We use show function as fast.show(), it shows the saved data on LEDs.
 If I need to glow 5000 LEDs with same data saved for first 100 LEDs. I don't want to write as #define NUM_LEDS 5000 because it consumes more space from micro-controller. How I can achieve this

jay1384, 2017-06-16:
can you please let me know the solution, Thanks in advance

marmilicious, 2017-06-16:
@jay1384  I don't think there's really a directly way to do that with FastLED.  But data output can be split/duplidated and sent to multiple strips, so you might be able to do something like set NUM_LEDS = 500, process your colors for the first 100 and copy it 4 times to the rest of NUM_LEDS.   Then when outputting split/duplicate your data line 10 times to feed the other 500 pixel strips.
(Maybe you could set NUM_LEDS to 100 and have strips of only 100 pixels, but then you'd need to split your data line 50 times which may or may not work so well.  I don't know.)

The github here is usually just used for bug reports.  Ask your question the G+ group and you might get more responses.  http://fastled.io/+  Include as much info about your setup and what you want to achieve as possible.

MFornander, 2017-06-16:
Couldn't you just instantiate a bunch of controllers with the same CRGB buffer?

_Sent from my Huawei Nexus 6P using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

jay1384, 2017-06-19:
Thanks a lot.
As you said feed same data to other 500 pixel strip, but In my case I am using only one strip with 5000 LEDs. So I am not very much clear about your idea. but thanks for support. I have also put the same discussion on G+ group.

marmilicious, 2017-06-19:
I updated my previous comment to hopefully be a bit more clear.  My idea was you could have ten strips with 500 pixels each (for a grand total of 5000 pixels).  The duplicated data signal would go to the other nine strips that each have 500 pixels.

gnkarn, 2017-06-19:
if you are using the leds in parallel all the time ( same animation on each 500 leds segment ) , then it may much more efficient , to split your strip, it will be also very chip as you only need a signal "booster" chip for the data  signal .

jay1384, 2017-06-19:
I dont want to split in to strips , I have single strip with 5000 LEDs.

yesyesuk, 2017-06-19:
You could fork the library and modify it so that it sends the same data 50 times.

jay1384, 2017-06-19:
I need to use only one GPIO, because I have single strip and not multiple
strips,I don't want to cut down strips in to 50 different strips. because
if i made 50 strips those will be at far distances then providing them data
with gpio is much more difficult , as long distance strength will decrease.

jay1384, 2017-06-19:
Exactly thats what i want to do, I tried number of times but I am unable to
achieve the same.

the-magister, 2017-06-19:
If you must save the memory, then you're tearing into the library to do a
replicated send.  That sounds very painful.

Alternately, you can spend $20 on a better microcontroller with more
memory.  Just because something can be done (read: re-writing the library
for your one-off case) it doesn't mean it should be done.

Without parallel outputs, an update to 5000 LEDs could lead to a very slow
frame rate.  For example, WS2811 has a per-LED write speed of 30 us.  With
5000 LEDS, that's one update every 0.15 seconds, or 6.67 updates per second
(i.e. 6 fps).  APA102 could possibly bail you out, but those are quite
expensive.

Given the scale of your installation, you should consider the OctoWS2811
daughterboard with a Teensy 3.1.  This makes your memory problem go away
and would increase your frame rate by a factor of 8x.

closed 2017-08-16 by focalintent