FastLED/FastLED #472: New WorldStar WS2812B strips

elliottkember, 2017-07-21:
New WS2812B strips are dropping. Annoyingly they're still labeled as WS2812B but they don't light up with the current version of FastLED I'm using. There's a small visible difference in the shape of the controller IC which was rectangular and is now square:

https://blog.adafruit.com/2017/05/03/psa-the-ws2812b-rgb-led-has-been-revised-will-require-code-tweak/
https://blog.particle.io/2017/05/11/ws2812b-neopixels-are-about-to-change/

Adafruit library has been patched here: https://github.com/adafruit/Adafruit_NeoPixel/commit/ec1614ad8f8ce54f5928cfa1e08e6e7b3b958faf something to do with changing a 50ms wait time to 300ms. 

It seems as though we can change this and have the code work with both versions(?) and I'm happy to try it with the strip I have here but I have no idea where to change it. Any help would be appreciated!!

jasoncoon, 2017-07-21:
Related converstations:

* https://github.com/FastLED/FastLED/issues/465
* https://plus.google.com/+AdamHaile/posts/YpriDmfew7u

elliottkember, 2017-07-21:
Thanks Jason! I should clarify that my strips don't light up or flicker at all, they just stay black. I doubt it's a faulty strip but I wouldn't rule that out.

Philxu007, 2017-08-03:
wow,
[ws2812b ](http://addressable-led.com/)is going to  the same layout as the [sk6812 led ](http://addressable-led.com/Products/sk6812-5050-rgb-led.html)

Varjohaltia, 2017-08-04:
I just bought some Adafruit 16 LED Neopixel rings (Product ID: 1463) and they don't work right. They light up, but the colors are weird and the demoreel demos basically leave them on at almost white. The fading doesn't work at all. Swapping in the current Adafruit demo sketch works fine, so it doesn't appear to be my hardware. Sounds like a similar issue.

Helpfully Adafruit says "May ship with either WS2812B or SK6812-based LEDs. They are the same brightness, color and protocol." I haven't been able to tell which I have.

Side-by-side images, my apologies for the atrocious quality.

[image]
[image]