FastLED/FastLED #475: FastLED + OctoWS2811 + Teensy 3.6 fails to build

LeonhardScheck, 2017-08-06:
While trying to use FastLED in combination with the OctoWS2811 library and a Teensy 3.6 I ran into problems:

- examples/Blink runs without problems on the Teensy 3.6
- examples/Multiple/OctoWS2811Demo runs without problems on the Teensy 3.2.
- examples/Multiple/OctoWS2811Demo fails to build for the Teensy 3.6

The build error message is 
```
In file included from src/main.cpp:3:0:
lib/FastLED/FastLED.h: In static member function 'static CLEDController& CFastLED::addLeds(CRGB*, int, int)':
lib/FastLED/FastLED.h:357:30: error: 'COctoWS2811Controller' does not name a type
case OCTOWS2811: { static COctoWS2811Controller<RGB_ORDER,WS2811_800kHz> controller; return addLeds(&controller, data, nLedsOrOf
fset, nLedsIfOffset); }
^
lib/FastLED/FastLED.h:357:105: error: 'controller' was not declared in this scope
case OCTOWS2811: { static COctoWS2811Controller<RGB_ORDER,WS2811_800kHz> controller; return addLeds(&controller, data, nLedsOrOf
fset, nLedsIfOffset); }
^
```

I am using the current master branch repository versions of OctoWS2811 and FastLED and PlatformIO 2.0.

focalintent, 2017-08-06:
The octows2811 controller is currently only supported on the teensy 3/3.1/3.2.

sumotoy, 2017-08-16:
> The octows2811 controller is currently only supported on the teensy 3/3.1/3.2.
It's compatible, looks here https://www.pjrc.com/teensy/td_libs_OctoWS2811.html

kcouck, 2017-08-16:
@sumotoy it's supported by the Teensy OctoWS software, but it isn't enabled for FastLED for Teensy 3.5/3.6

focalintent, 2017-08-16:
There is preliminary 3.6 support in place that someone contributed code to.  I'm making a quick change that should hopefully enable the OctoWS2811 driver for the 3.6 as well right now.

sumotoy, 2017-08-16:
I've tested time ago the [kcouck]https://github.com/kcouck(url) K66 addon and compiled without any error. I also uncommented the octows2811_controller.h and smartmatrix_t3.h, again everithing compiled without error. I can test with a 3.6 tomorrow and see what's happen.

kcouck, 2017-08-16:
Yeah, I couldn't test it, so didn't enable it... in theory the same applies for SPI support, but that was harder to take out :)
As I said to @focalintent in the 3.6 pull request, there's probably a lot of redundant code now between the k20 and k66 dirs (not mentioning k64 if support for Teensy 3.5 is added (which is 99% the same as k66)

commit 7a064b9f (2017-08-16): Fix #475, add support for the octows2811 adapter
closed 2017-08-16 by focalintent