FastLED/FastLED #477: WS2812B flickers. WS2812 did'nt

hmiez, 2017-08-14:
WS2812B flicker, WS2812 not.

I am facing a big issue when switching from WS2812 to WS2812B !

My configuration :
•	IOT application, 6000 loc, including FASTLED 3.1 and AsyncWEBServer
•	WS2812, 170 LEDS, ESP8266 on NodeMCU or ESP03, Strip cmd on D6 - GPIO12 (but also tested and working  on other GPIO’s), strip refreshed at 100Hz, interrupts never disabled, WIFI and LEDS working in parrallel. 
FastLED pragma « All SPI access will default to bitbanged output » at compile time. I nerver succeeded to have FastLED using SPI output (D7 – RXD2)
•	All this worked smooth and stable for months with WS2812 strips.

Now I want to build a dozen device, WS2812 is no longer available so I ordered WS2812B at AliBaba and here, I have a problem !

With WS2812 ledstrip, all was doing well (no esp resets, nice display, no flickering).
When I replaced 2812 by 2812B (nothing else changed) many flickering occur (most often an area in the strip that is supposed to be off lits for a very short amount of time, probably between 2 FastLED show). 
I tried to change #define CHIPSET from 2812 to 2812B or 2813 and did not see any noticeable difference.
I first suspected a quality issue with the strip, but the result was the same with other strips from other suppliers. 

So, bad luck : it seems that the problem is really the SW and not 2812B strip. 

Just to confirm, I tried 2 SW simple configurations with the same HW:

•	NeoPixel Bus in several configurations still flicker but..
I found that using data pin 4 – GPIO02 and  explicitly using the UART hardware behind it, I HAD NO FLICKERING on WS2812B !
NeoPixelBus<NeoGrbFeature, NeoEsp8266Uart800KbpsMethod> strip(PixelCount);
So, this damned ledstrip can really work !
•	On the same data pin, with FastLED Fire2012WithPalette is perfect with 2812 but flickers with 2812B (FastLED still bitbangs outputs on data pin 4)

Seems that it has nothing to do with HW, the only thing I can suspect is SW timing (as often with ledstrips).

Now I have lots of PCB’s which I cannot use and I am not very motivated to start fishing obsolete WS2812 on the internet…

Any idea to solve this with FastLED* ?

* which I appreciate a lot. 
By the way, lots of thanks to the authors of FastLED who did a great great and useful work.

todbot, 2017-08-17:
How are you powering the LEDs?  If this is ESP8266 with 3.3V GPIO, are you 3.3V->5V level-shifting the data line to be compliant with the 5V logic of the WS2812 LEDs?

focalintent, 2017-08-17:
The problem is likely the intersection of interrupts and timing and wifi on the eps8266 - a couple things to try is either:

```
#define FASTLED_ALLOW_INTERRUPTS 0
#include <FastLED.h>
```

Or 

```
#define FASTLED_INTERRUPT_RETRY_COUNT 1
#include <FastLED.h>
```

hmiez, 2017-08-18:
Thank you, focalintent for your answer.
I already tried #define FASTLED_ALLOW_INTERRUPTS 0 and it did not change the flickering, even when wifi was disabled (I really did a unit test, not in my app)
I will try #define FASTLED_INTERRUPT_RETRY_COUNT 1 in coming days and will keep you informed.
What is surprising to me is that this problem seems not to be known from other users (I beleive many people may use FastLed together with WS2812B). Any idea ?

focalintent, 2017-08-18:
There’ve been multiple issues filed here and a handful of conversations on g+ as well regarding the timing of WS2812B’s and interrupts, and the esp8266.  Honestly stuff like this is why i’ve mostly moved to APA102 for my own projects.  The 2812’s timings are frequently a source of problems.

hmiez, 2017-08-19:
Thank for sharing your experience, focalintent !
I will probably make the same choice, but have to check the prices before, it seems that APA102 are somewhat more expensive.

closed 2017-08-19 by hmiez