FastLED/FastLED #480: ESP8266 with APA102 Buffered/Delayed Output Behavior

devbergey, 2017-08-19:
Hello all,
I'm finding some interesting behavior now using an NodeMCU ESP8266 board and APA102 LEDs. Previously, I was using a Teensy and getting output to my LEDs, however, when I switched over to an EPS8266 board, my output seems to be buffered. I will give my CRGB array data and on my FastLED.show() command, I get no output. It's not until the next FastLED.show() command do I get output onto my LEDs.

Has anyone run into this issue? I've seen other people in the community (and youtube) successfully using FastLED with the ESP8266 and APA102s.

focalintent, 2017-08-20:
This actually sounds like your leds are behaving as if they are SK9822's, not APA102's - try switching the led type to that.

devbergey, 2017-08-30:
Yes, thank you, that resolved the issue. I thought I was using the APA102s.

closed 2017-08-30 by devbergey