FastLED/FastLED #485: Makefile Compilation

NicoHood, 2017-08-23:
Hi,
I try to compile FastLED based on a [DMBS makefile](https://github.com/abcminiuser/dmbs) for an atmega32u4 (no Arduino environment).

I got the code compiling with a few tricks, but I have not tested it, and only included the FestLED.h file, nothing more. It would be nice if you could give me some guidiance what is required to use FastLED with a pure makefile. I am willing to write a DMBS makefile (and patches via PR, if required) so others can use that too.

My findings are based on FastLED 3.1.6

in pixelset.h abs() was undefined. I added used:
```
#include "FastLED.h"
#ifndef ARDUINO
#include "stdlib.h"
#endif
```

FastLED requires some Arduino functions for pinmapping (as a fallback). I disabled it, and then no pinMode functions (etc.) were not required anymore. Is this the correct way? The pins for an atmega 32u4 should be defined already, I am using the same layout.
```
CC_FLAGS     =  -D FASTLED_NO_PINMAP
```

I also upgraded to c++11 (something I found on some website, more compiler flags might be required):
```
CPP_STANDARD = gnu++11
CPP_FLAGS = -fno-threadsafe-statics
```

And finally I added some more code to suppress some more errors. I know that I need to reimplement micros() later.
```
typedef uint8_t byte;
typedef bool boolean;

static uint32_t micros(){
    return 0;
}
```

Does anyone has more experience on using FastLED with AVRs but via makefile? Any (older) examples maybe?

NicoHood, 2017-08-23:
Thanks! This reduced the modifications.

Those are the changes I made directly to fastled. Maybe those can be fixed upstream:
https://github.com/FastLED/FastLED/compare/master...NicoHood:makefile?expand=1

I've added this special header via `-include file.h` compiler flag:
```cpp
#pragma once

#include <stdint.h>
#include <stdbool.h>
#include <stddef.h>

typedef uint8_t byte;
typedef bool boolean;

static uint32_t micros(){
    return 0;
}

static uint32_t millis(){
    return 0;
}

static void delay(uint32_t delay){

}
```

And the extra makefile options:
```make
SRC          = $(TARGET).cpp $(AVR_LIB_PATH)/FastLED/FastLED.cpp $(AVR_LIB_PATH)/FastLED/power_mgt.cpp $(AVR_LIB_PATH)/FastLED/noise.cpp $(AVR_LIB_PATH)/FastLED/wiring.cpp $(AVR_LIB_PATH)/FastLED/lib8tion.cpp $(AVR_LIB_PATH)/FastLED/hsv2rgb.cpp $(AVR_LIB_PATH)/FastLED/colorutils.cpp $(AVR_LIB_PATH)/FastLED/bitswap.cpp
CC_FLAGS     = -I $(AVR_LIB_PATH)/FastLED -include $(AVR_LIB_PATH)/core.h -D FASTLED_NO_PINMAP
CPP_STANDARD = gnu++11
CPP_FLAGS += -fpermissive -fno-exceptions -fno-threadsafe-statics 
#-fno-devirtualize
CPP_FLAGS += -DNEED_CXX_BITS
```

NicoHood, 2017-08-23:
Can't you use _delay_ms instead of delay? Maybe calling delay_micros(1000) might make more sense in being more precise.
http://www.atmel.com/webdoc/avrlibcreferencemanual/group__util__delay_1gad22e7a36b80e2f917324dc43a425e9d3.html

focalintent, 2017-08-23:
I’m using delay because that’s what’s on the arduino platform and that is what the largest majority of library users by far is using.  I’m not going to drop to platform specific code outside of the platforms directory.  

Moving to stdint types is an ongoing thing that happens with other work, or rather will be once I get back to actively working on the library.

focalintent, 2017-08-23:
Also on arduino delay calls yield which I believe delayMicroseconds does not.

NicoHood, 2017-08-23:
If I am not wrong those commits should be easy to merge:
https://github.com/FastLED/FastLED/commit/f360030e991b93b6cc5e69c9b92cd894e923db9b
https://github.com/FastLED/FastLED/commit/d73ed0d1bec483b3ef4ec84d22350205a53da22f

About the other things:
Maybe we could add `ifndef ARDUINO` around those which should get rid of those problems and the code should work for both?

I currently have no led hardware. Would be interesting to test my output. Maybe we could add makefile support with very few fixes.

focalintent, 2017-08-23:
The type updates I’ll take — the tossing ifdef stuff around is not something I’m willing to take blindly without testing myself, especially since some of the things are things that will need local replacement, the compiler errors will be useful for that for folks building outside the arduino environment.  

And again - while I’ve done work to make building outside of arduino easier to do it is not something I want to claim is supported nor is it something that I want to actively support.

NicoHood, 2017-08-23:
Thanks! Sure, I totally understand. I also need to test this myself. It might take some time until I get my hands on real led hardware. I will report back, when I got more updates :)

ladislas, 2017-08-31:
Hi there! 
First thanks a lot @focalintent for the amazing work! And thank you @NicoHood for all the information you provided.

I was able to compile FastLED using the [Arduino-Makefile](https://github.com/sudar/Arduino-Makefile).

Furthermore, I did not need to set `-DNEED_CXX_BITS` in my Makefile or in the code. Is that normal? Setting it will cause an error and complain that `extern "C" void __cxa_pure_virtual (void) {}`
 is defined multiple times.

focalintent, 2017-08-31:
It depends on the platform and target, which is why it is something that isn’t on all the time by default.

NicoHood, 2017-08-31:
@ladislas I think Arduino-makefile is just another tools similar to arduino-builder. You do not need to use the IDE (GUI), and can use a commandline tool instead. However this does not get rid of the whole Arduino core (which my aim is).

And thatswhy the Arduino code gets pulled in, you have the function defined multiple times and no real improvement from my point of view. In your case I'd rather directly use arduino-builder.

@focalintent I got hardware (NEOPIXEL) to test now. I will prepare a PR once I got all fixes applied. I will try to make it generally work, so nothing else breaks. The changes should be very small.

ladislas, 2017-09-01:
@NicoHood do you have a repo with your project somewhere? I'd love to take a look and maybe help.

I've just managed to make FastLED Neopixel work with [ChRT](https://github.com/greiman/ChRt) ([ChibiOS/RT for Arduino](http://www.chibios.org/dokuwiki/doku.php).

I'm waiting for my APA102 to arrive to do a comparison.

NicoHood, 2017-09-01:
Oh yes, I did not mention yet:
https://github.com/NicoHood/avr

The makefile works. Neopixel Leds are working (tested!) and the other chipsets seem to compile. If you want to reproduce it you must make sure that you use my fastled fork, as well as the dmbs fork as described in the readme. I hope those fixes get applied soon :) I will prepare a PR for FastLED.

closed 2019-01-20 by focalintent