FastLED/FastLED #489: FastLED 3.1.6 + Arduino 1.8.3 not working with NodeMCU ESP-12?

timfield, 2017-08-30:
Build and deploy FastLED Blink demo, using DATA_PIN D4 and WS2812B strand (also tested with APA102 without success). I added Println() commands to show app is cycling through code, but LED[0] doesn't budge. Am powering LED with separate 5v supply. Exact same hardware setup works fine with Adafruit NeoPixel strandtest.

I normally use Wemos D1 mini for my ESP8266 FastLED projects, but needed extra pins of NodeMCU... and this all works fine with Wemos. I successfully used NodeMCU (probably a year ago), so this is new issue. Also, if I try to use DATA_PIN D5 or greater, I get "Invalid Pin" assertion (even though these are indeed valid on NodeMCU).

Are others using NodeMCU successfully with current 3.1.6 release? Any suggestions for me to try?

Thanks!!

frhoffer, 2017-08-31:
I was able to get my NodeMCU ESP-12 board to work with Arduino 1.8.3 and FastLED 3.1.6. I had to add the define FASTLED_ESP8266_RAW_PIN_ORDER. I used D8 and WS2811 but probably would work with D4 and your WS2812B strand. Here's my sample code:

#define FASTLED_ESP8266_RAW_PIN_ORDER
#include "FastLED.h"
#define NUM_LEDS 50
#define PIN D8
#define COLOR_ORDER RGB

CRGB leds[NUM_LEDS];
  void setup() { FastLED.addLeds<WS2811, PIN>(leds, NUM_LEDS); }
  void loop() {
      for(int x = 0; x < NUM_LEDS; x++){
          leds[x] = CRGB(255,50,0);
      FastLED.show();
      delay(50); 
      leds[x] = CRGB(0,0,0);
    }
  }

eboston, 2017-09-01:
Could we get this define added to the documentation?  I was having this issue too and the define fixed it, but I haven't noticed it anywhere in the documentation.

closed 2017-10-06 by timfield

huangguozhen, 2018-05-21:
@timfield 
i use FASTLED_ESP8266_RAW_PIN_ORDER, but it don't work right.

dmak, 2019-10-07:
Seems to be the same as issue #761. In my case (LOLIN NodeMCU V3 or Wemos D1 mini) the following has solved the issue:
```
#define FASTLED_ESP8266_RAW_PIN_ORDER

#include <FastLED.h>
```

Smuno, 2019-12-21:
In the case of LOLIN NodeMCU V3 can only use GPIO13 and GPIO14 (d7 and d5) and must use:
`#define FASTLED_ESP8266_RAW_PIN_ORDER
#include <FastLED.h>`

Info from [this awesome guy](https://medium.com/@cr0wg4n/controlando-tiras-led-con-fastled-en-nodemcu-esp-8266-f20445735a22)