FastLED/FastLED #491: Resetting after .show() on ESP32

teldredge, 2017-09-03:
I'm using a WeMos Lolin32 and when I try to run the blink program it seems to reset the board.

I'm getting this output on serial:

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:588
load:0x40078000,len:0
load:0x40078000,len:10472
entry 0x40078a28
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c:1400 (vTaskDelay)- assert failed!
ets Jun  8 2016 00:22:57

sashasklar, 2017-09-18:
Are you sure it's on the .show() and not on the .delay()? I had a similar issue and switching to the native arduino delay from fastLED.delay resolved it.

sashasklar, 2017-09-19:
I am able to reproduce this with larger pixel sets. It looks like the culprit is the esp32 watchdog timer. I was able to work around it by making smaller updates with show and wrapping the calls with yield().

hobzcalvin, 2017-12-27:
I'm seeing similar, though my output is:
```
rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, { garbled mess }
```

There's probably an easy way to disable the watchdog timers, right?

vseven, 2017-12-31:
I'm having the same issue with the standard NeoPixel library, cutting it down to much smaller updates seems to work and also if I let it sit for a while.  It doesn't make much sense.  @sashasklar  - Can you show a example or just post the code you are using?

hobzcalvin, 2018-03-27:
I'd love to tackle this bug now that @alamut has a simple reproducing example but I don't really know where to begin. Could @samguyer @focalintent @tullo-x86 provide any thoughts on what might be causing the issue? Is this a common thing when an ESP32 runs out of some sort of memory, etc.?

focalintent, 2018-03-27:
It’s the watchdog timer - it’s something these systems have to catch the possibility of an infinite loop - and if that happens it resets the MCU.  The question is, what is writing out led data blocking that’s causing the timer to fire - on some systems the timer gets reset every time loop() exits.  On others, every time yield is called.  On still others, it’s enough for certain interrupt handlers to be allowed to fire.

VioletGiraffe, 2018-04-18:
The same problem with the [following sketch](https://gist.github.com/VioletGiraffe/f25e048f14a21624547d195db57466e8) and WS2812 matrix, doesn't get simpler than that. Note that I even commented out `delay()`, to no avail.

This is with the latest FastLED from the repo and the latest ESP32 Arduino platform from Espressif repo.
Here's the diagnostic output:

> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:1
> load:0x3fff0018,len:4
> load:0x3fff001c,len:956
> load:0x40078000,len:0
> load:0x40078000,len:13256
> entry 0x40078a90
> Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
> Core 1 register dump:
> PC      : 0x400d0b38  PS      : 0x00060134  A0      : 0x800d0c41  A1      : 0x3ffca560  
> A2      : 0x3ffca594  A3      : 0x0523ce95  A4      : 0x00000002  A5      : 0x0000003b  
> A6      : 0x00040000  A7      : 0x000000d1  A8      : 0x80000000  A9      : 0x3ffc22d5  
> A10     : 0x3ff44004  A11     : 0x00040000  A12     : 0x0523cf01  A13     : 0x3ff44004  
> A14     : 0x00040000  A15     : 0x0000012b  SAR     : 0x0000000e  EXCCAUSE: 0x00000006  
> EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  
> 
> Backtrace: 0x400d0b38:0x3ffca560 0x400d0c3e:0x3ffca590 0x400d0d31:0x3ffca5d0 0x400d0e59:0x3ffca610 0x400d09a5:0x3ffca650 0x400e351d:0x3ffca670
> 
> Core 0 register dump:
> PC      : 0x400d2fbe  PS      : 0x00060534  A0      : 0x800856d1  A1      : 0x3ffc6d90  
> A2      : 0x00000008  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x3ffc73d4  
> A6      : 0x00000000  A7      : 0x00000001  A8      : 0x3ffc2984  A9      : 0x3ffc2968  
> A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x00000001  
> A14     : 0x00060021  A15     : 0x00060c23  SAR     : 0x00000000  EXCCAUSE: 0x00000006  
> EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
> 
> Backtrace: 0x400d2fbe:0x3ffc6d90 0x400856ce:0x3ffc6db0
> 
> Rebooting...

samguyer, 2018-04-18:
@violetgiraffe Have you tried my branch?

VioletGiraffe, 2018-04-18:
@samguyer Thanks for getting back to me. I thought the relevant changes had already been merged back into the original FastLED library, but apparently not.
Tried the master branch of your FastLED fork. It's much better, but now i'm getting another problem: only the first 3 LEDs of my 8x8 matrix are lit. Previously, all the LEDs were blinking, so I know the matrix is not defective.

samguyer, 2018-04-18:
@VioletGiraffe Can you post your code?

VioletGiraffe, 2018-04-18:
@samguyer Sure (and I already have): 
```
#include <FastLED.h>
#define DATA_PIN 19

constexpr auto NUM_LEDS = 64;
CRGB leds[NUM_LEDS];

void setup() {
  FastLED.addLeds<NEOPIXEL, DATA_PIN>(leds, NUM_LEDS);

  for (auto i = 0; i < NUM_LEDS; ++i)
     leds[i] = CRGB::Black;
}

void loop() {
  for (int dot = 0; dot < NUM_LEDS; ++dot) { 
      leds[dot].setRGB(255, 200, 0);
      FastLED.show();
      // clear this led for the next time around the loop
      leds[dot] = CRGB::Black;
      delay(30);
  }
}
```

samguyer, 2018-04-18:
@VioletGiraffe Hmmm. I don't see anything obviously wrong. Can you try something for me? Put the code that sets the dot to black *after* the delay.

VioletGiraffe, 2018-04-18:
@samguyer That did it! What are you thinking, got an idea?

samguyer, 2018-04-18:
@violetgiraffe Great! Ok, I think I'm being a little aggressive about the asynchronous output, so what's happening is that your code changes the LED to black while my code is still sending the data. That should not be able to happen, so I need to look into the synchronization.

guidoffm, 2018-08-08:
Experienced the same problem. Continuosly SW_CPU_RESET on show()

samguyer, 2018-08-08:
@VioletGiraffe I have made some substantial changes to my branch in the last couple of weeks. I discovered that the heap allocation I was doing was causing memory to become very badly fragmented. That's fixed now.

VioletGiraffe, 2018-08-08:
Thanks, great news (and thank you for developing this fork). I'll give it a try as soon as I'm back to working on my ESP32 project.

guidoffm, 2018-08-08:
```
#include "FastLED.h"
#define DATA_PIN 14
#define NUM_LEDS 150

CRGB leds[NUM_LEDS];

void setup() {
  Serial.begin(115200);

  Serial.println("addLeds");
  FastLED.addLeds<WS2812B, DATA_PIN, RGB>(leds, NUM_LEDS);
}

void loop() {
    for (int i=0; i<256;i++){
      fill_rainbow( &(leds[0]), NUM_LEDS, i /*starting hue*/);
      FastLED.show(); 
      delay(100);
    }
}
```

closed 2019-01-20 by focalintent