FastLED/FastLED #492: 32u4 + Neopixel not working

NicoHood, 2017-09-04:
```cpp
#include "FastLED.h"
CRGB leds[1];
void setup() {
  FastLED.addLeds<NEOPIXEL, 11>(leds, 1);
}

void loop() {
  leds[0] = CRGB::Red;
  FastLED.show();
  delay(500);
  leds[0] = CRGB::Black;
  FastLED.show();
  delay(500);
}
```

This is not functional with Arduino Leonardo. The same sketch is working on my Uno. I made sure the pin is correct, by blinking an led on the same pin also. Using IDE 1.8.4 with archlinux. Tested with packaged avr-gcc (4.9) and upstream avr-gcc (7.1). I have no other chipsets to test. Issu occurs with master branch and 3.1.6

closed 2019-01-20 by focalintent

NicoHood, 2019-01-20:
Was the issue fixed?

focalintent, 2019-01-20:
I’m not sure what was causing the issue you reported - but I know I was just running similar code on a 32u4 with no issues

irl-ops, 2021-03-11:
I had the same problem, with the code below if you do not define the pins for FastLED to use then you never get "loop" printed  in the serial monitor and the LED does not blink.  If you do tell FastLED which pins to use the code works fine.  See code below:

```
#include <Arduino.h>
#include <FastLED.h>

#define NUM_LEDS 9
CRGB leds[NUM_LEDS];

void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN,LOW);
  Serial.begin(9600);

  //         FastLED.addLeds<DOTSTAR, BGR>(leds, NUM_LEDS);  // this does not work, nothing on serial monitor, LED does not blink
  FastLED.addLeds<DOTSTAR, SCK, MOSI, BGR>(leds, NUM_LEDS);  // serial monitor works, prints "loop", LED blinks
  FastLED.setBrightness(122);
  FastLED.clear();
  fill_solid(leds, NUM_LEDS, CRGB::Red);
  FastLED.show();

  }

void loop()
{
  Serial.println("loop");
  digitalWrite(LED_BUILTIN, !digitalRead(LED_BUILTIN));
  delay(1000);
}
```

jsheedy, 2022-04-22:
I had problems using hardware SPI on the 32u4 until I followed [this advice to set PB0 to Output](https://arduino.stackexchange.com/questions/3719/arduino-hardware-spi-hangs-bit-bang-does-not) in `setup()` via
```
pinMode(17, OUTPUT);
```

With that in place my DotStar LEDs work with hardware SPI