FastLED/FastLED #496: Cant get the calls to work with NodeMCU

rhamblen, 2017-09-12:
I have a strip of ws2811 lights and when I use McLighting, it all works.
However, I want to build a matrix!  So I did some testing with both FastLED and NeoPixel and I cant get the calls to work, even using the examples calling for the strip to be turned on.
I am using the the same hardware configuration and pin 5 which is GPIO5 (labeled D1) to no avail!

I assumed it was the setup and ive tried both...
  LEDS.addLeds<WS2811, DATA_PIN, BRG>(leds, NUM_LEDS);
  FastLED.addLeds<WS2811, DATA_PIN, BRG>(leds, NUM_LEDS);

I've added debug statements and can see it gets into the loop to call the light...
    leds[i] = CRGB::Red;
    FastLED.show();

I've also done the 
  #define FASTLED_ESP8266_NODEMCU_PIN_ORDER

I dont have a scope to put on the pin but I know there is no signal being sent.. the voltage is low when using McLighting the voltage is high when signal sent.

If it didnt work with McLightnight I would say it was voltage level on the data line, but this appears to be something about the setup as I dont think it is even sending data.

I've tried a NodeMCU v1.0 and v2.0 and Ive tried different pins too.

I'm at a loss!  Help!!!

focalintent, 2017-09-12:
Have you tried one of the other esp8266 pin orders?   Have you tried any of the other puns?

rhamblen, 2017-09-12:
tried other pins and also tried 
#define FASTLED_ESP8266_RAW_PIN_ORDER
as per earlier post on issue below
tried different pin orders..

```
#define FASTLED_ESP8266_RAW_PIN_ORDER
//#define FASTLED_ESP8266_NODEMCU_PIN_ORDER
//#define FASTLED_ESP8266_D1_PIN_ORDER

#include "FastLED.h"
#define NUM_LEDS 50
#define PIN D8
//#define COLOR_ORDER BGR

CRGB leds[NUM_LEDS];
  void setup() {
    Serial.begin(115200); 
    delay(5000);
    Serial.println("Serial initalized!\n");
    Serial.println("         1         2         3         4         5         6         7");
    Serial.println("1234567890123456789012345678901234567890123456789012345678901234567890");

    //FastLED.addLeds<WS2811, PIN>(leds, NUM_LEDS); 
    FastLED.addLeds<WS2811, PIN, BRG>(leds, NUM_LEDS);
  }

  void loop() {
    for(int x = 0; x < NUM_LEDS; x++){
      leds[x] = CRGB(255,50,0);
      FastLED.show();
      Serial.print(".");
      delay(50);
      leds[x] = CRGB(0,0,0);
    }
    Serial.println("/");
  } 

```

focalintent, 2017-09-12:
As an aside, don’t use the Dx pin identifiers - just use the PIN number (there’s a reason why all the examples use just numbers and not D8, for example).

rhamblen, 2017-09-12:
so I think I have it sorted.
I swapped the LED strips over and it works.  It appears the first strip is suseptable to noise!

closed 2017-09-12 by rhamblen