FastLED/FastLED #499: using FastLED while producing sounds

nitbeat, 2017-09-14:
Hi,

I noticed this issue:
https://github.com/FastLED/FastLED/issues/208

It's from 2015, so I'm just hoping for a current solution using FastLED while producing sounds.
If not Tone() maybe someone knows a working alternative ?

How do you make sounds while using FastLED 

Arduino Nano V3 - Atmegs328

focalintent, 2017-09-14:
Use leds that don't require disabling interrupts - for example APA102's.  Also - as mentioned in the issue that you linked to - for general "how do I" questions, please use the g+ group.

> (And in general, we use this issue tracker for bugs. For questions and how-to help, try the Google Plus group for FastLED: http://fastled.io/+ )

focalintent, 2017-09-14:
Alternatively, use a platform, like the teensy 3.2, that allows running interrupts while writing out WS2812 led data.

closed 2017-09-14 by focalintent

nitbeat, 2017-09-15:
What is different in teensy 3.2 ? is it the MCU being used there, or the platform itself ?

focalintent, 2017-09-15:
The difference is the MCU - the teensy 3.2 is a 32-bit arm cpu running at 72-168Mhz, while the nano is an 8-bit avr cpu running at 16Mhz.  There's simply not enough spare cpu cycles on the avr to drive the WS2812 leds _and_ allow interrupts to process.