FastLED/FastLED #500: Adafruit Circuit Playground Express - SAMD21 w/different pins

eduncan911, 2017-09-16:
Hello:

We have a project trying to use the FastLED code on the SAMD21 with different pinouts.  It works with the older version of our device, the Adafruit Circuit Playground that has the 8-bit ATmega32u4.  

https://learn.adafruit.com/circuit-playground-seashell-pendant/code-1

Any help with the below would be appreciated.  Using the latest Arduino

The newer boards have an 32-bit SAMD21.  Same pins as the older versions, and we are using the same code that attempts to change the INPUT.  Just the CPU arch changes, and more sensors/inputs are available.  Here's the errors we have:

https://forums.adafruit.com/viewtopic.php?f=58&t=122310&p=617198 (pasted below)

```
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/eric/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/eric/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/eric/Documents/Arduino/libraries -fqbn=arduino:samd:adafruit_circuitplayground_m0 -vid-pid=0X239A_0X8018 -ide-version=10804 -build-path /var/folders/fk/kyjxtwcj7ync9w2y2l6573kh0000gn/T/arduino_build_736605 -warnings=none -build-cache /var/folders/fk/kyjxtwcj7ync9w2y2l6573kh0000gn/T/arduino_cache_16809 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=/Users/eric/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.2.0 -prefs=runtime.tools.arm-none-eabi-gcc.path=/Users/eric/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 -prefs=runtime.tools.openocd.path=/Users/eric/Library/Arduino15/packages/arduino/tools/openocd/0.9.0-arduino6-static -prefs=runtime.tools.bossac.path=/Users/eric/Library/Arduino15/packages/arduino/tools/bossac/1.7.0 -prefs=runtime.tools.CMSIS.path=/Users/eric/Library/Arduino15/packages/arduino/tools/CMSIS/4.5.0 -prefs=runtime.tools.CMSIS-Atmel.path=/Users/eric/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0 -verbose /Users/eric/Documents/Arduino/neckless/neckless.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/eric/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/eric/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/eric/Documents/Arduino/libraries -fqbn=arduino:samd:adafruit_circuitplayground_m0 -vid-pid=0X239A_0X8018 -ide-version=10804 -build-path /var/folders/fk/kyjxtwcj7ync9w2y2l6573kh0000gn/T/arduino_build_736605 -warnings=none -build-cache /var/folders/fk/kyjxtwcj7ync9w2y2l6573kh0000gn/T/arduino_cache_16809 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=/Users/eric/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.2.0 -prefs=runtime.tools.arm-none-eabi-gcc.path=/Users/eric/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 -prefs=runtime.tools.openocd.path=/Users/eric/Library/Arduino15/packages/arduino/tools/openocd/0.9.0-arduino6-static -prefs=runtime.tools.bossac.path=/Users/eric/Library/Arduino15/packages/arduino/tools/bossac/1.7.0 -prefs=runtime.tools.CMSIS.path=/Users/eric/Library/Arduino15/packages/arduino/tools/CMSIS/4.5.0 -prefs=runtime.tools.CMSIS-Atmel.path=/Users/eric/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0 -verbose /Users/eric/Documents/Arduino/neckless/neckless.ino
Using board 'adafruit_circuitplayground_m0' from platform in folder: /Users/eric/Library/Arduino15/packages/arduino/hardware/samd/1.6.16
Using core 'arduino' from platform in folder: /Users/eric/Library/Arduino15/packages/arduino/hardware/samd/1.6.16
Detecting libraries used...
"/Users/eric/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10804 -DARDUINO_SAMD_CIRCUITPLAYGROUND_EXPRESS -DARDUINO_ARCH_SAMD  -DCRYSTALLESS -DARDUINO_SAMD_ZERO -D__SAMD21G18A__ -DUSB_VID=0x239A -DUSB_PID=0x8018 -DUSBCON '-DUSB_MANUFACTURER="Adafruit"' '-DUSB_PRODUCT="Circuit Playground Express"' "-I/Users/eric/Library/Arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/Users/eric/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/" "-I/Users/eric/Library/Arduino15/packages/arduino/hardware/samd/1.6.16/cores/arduino" "-I/Users/eric/Library/Arduino15/packages/arduino/hardware/samd/1.6.16/variants/circuitplay" "/var/folders/fk/kyjxtwcj7ync9w2y2l6573kh0000gn/T/arduino_build_736605/sketch/neckless.ino.cpp" -o "/dev/null"
"/Users/eric/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10804 -DARDUINO_SAMD_CIRCUITPLAYGROUND_EXPRESS -DARDUINO_ARCH_SAMD  -DCRYSTALLESS -DARDUINO_SAMD_ZERO -D__SAMD21G18A__ -DUSB_VID=0x239A -DUSB_PID=0x8018 -DUSBCON '-DUSB_MANUFACTURER="Adafruit"' '-DUSB_PRODUCT="Circuit Playground Express"' "-I/Users/eric/Library/Arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/Users/eric/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/" "-I/Users/eric/Library/Arduino15/packages/arduino/hardware/samd/1.6.16/cores/arduino" "-I/Users/eric/Library/Arduino15/packages/arduino/hardware/samd/1.6.16/variants/circuitplay" "-I/Applications/Arduino.app/Contents/Java/libraries/Adafruit_Circuit_Playground" "/var/folders/fk/kyjxtwcj7ync9w2y2l6573kh0000gn/T/arduino_build_736605/sketch/neckless.ino.cpp" -o "/dev/null"
"/Users/eric/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10804 -DARDUINO_SAMD_CIRCUITPLAYGROUND_EXPRESS -DARDUINO_ARCH_SAMD  -DCRYSTALLESS -DARDUINO_SAMD_ZERO -D__SAMD21G18A__ -DUSB_VID=0x239A -DUSB_PID=0x8018 -DUSBCON '-DUSB_MANUFACTURER="Adafruit"' '-DUSB_PRODUCT="Circuit Playground Express"' "-I/Users/eric/Library/Arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/Users/eric/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/" "-I/Users/eric/Library/Arduino15/packages/arduino/hardware/samd/1.6.16/cores/arduino" "-I/Users/eric/Library/Arduino15/packages/arduino/hardware/samd/1.6.16/variants/circuitplay" "-I/Applications/Arduino.app/Contents/Java/libraries/Adafruit_Circuit_Playground" "-I/Users/eric/Library/Arduino15/packages/arduino/hardware/samd/1.6.16/libraries/Wire" "/var/folders/fk/kyjxtwcj7ync9w2y2l6573kh0000gn/T/arduino_build_736605/sketch/neckless.ino.cpp" -o "/dev/null"
"/Users/eric/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10804 -DARDUINO_SAMD_CIRCUITPLAYGROUND_EXPRESS -DARDUINO_ARCH_SAMD  -DCRYSTALLESS -DARDUINO_SAMD_ZERO -D__SAMD21G18A__ -DUSB_VID=0x239A -DUSB_PID=0x8018 -DUSBCON '-DUSB_MANUFACTURER="Adafruit"' '-DUSB_PRODUCT="Circuit Playground Express"' "-I/Users/eric/Library/Arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/Users/eric/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/" "-I/Users/eric/Library/Arduino15/packages/arduino/hardware/samd/1.6.16/cores/arduino" "-I/Users/eric/Library/Arduino15/packages/arduino/hardware/samd/1.6.16/variants/circuitplay" "-I/Applications/Arduino.app/Contents/Java/libraries/Adafruit_Circuit_Playground" "-I/Users/eric/Library/Arduino15/packages/arduino/hardware/samd/1.6.16/libraries/Wire" "-I/Users/eric/Library/Ardu [... cut ...]
[=======                       ] 23% (128/544 pages)
[==========                    ] 35% (192/544 pages)
[==============                ] 47% (256/544 pages)
[=================             ] 58% (320/544 pages)
[=====================         ] 70% (384/544 pages)
[========================      ] 82% (448/544 pages)
[============================  ] 94% (512/544 pages)
[==============================] 100% (544/544 pages)
done in 0.220 seconds

Verify 34808 bytes of flash with checksum.
Verify successful
done in 0.049 seconds
CPU reset.
```

focalintent, 2017-09-16:
My guess is that the adafruit circuit playground has a different set of mappings between the CPU level GPIO register ports and pins than other D21 based platforms - so a new set of pin definitions would need to be made to line things up properly.

focalintent, 2017-09-16:
Note that the program is compiling and uploading - and it's probably doing what you want/expect it to - the output is occurring on a different pin than you were expecting.

eduncan911, 2017-09-17:
Thanks @focalintent !

You are correct.  It has different mappings (noticed from some I2C LCD examples I tried to get working and failed); as well as it is compiling and uploading, just using a different pin.

The thing is, this works with the PIN assignments on the older version of the board with the 8-bit CPU.  

With the newer CPU, I've been able to run all other code examples that uses custom pins and libraries in the Playground ecosystem - which all use the same PINs.

I saw your notes here in issue #393 and thought that perhaps I should force the CPU to use `ARDUINO_SAM_ZERO`.  The output posted above shows it using `ARDUINO_SAMD_ZERO`.  Given, I have no idea what the differences of these are.  Kind of looking for a quick fix.

Considering the PIN assignments in code between the older 8-bit and newer 32-bit are the same (cause the existing examples continue to work), and considering the entire thing does build and get uploaded - could it be possible that this FastLED library is not accepting a simple thing like which PIN to listen to, or that a definition for what PIN to use is different on newer CPUs?  

I'm wondering if assigning a PIN is different between the CPUs within FastLED's library.  And if that's the case, what would "PIN 17" on the old 8-bit `ATmega32u4` be on the newer 32-bit `SAMD21`?

```
// tell FastLEED all about the Circuit Playground's layout
 
#define CP_PIN      17   //circuit playground's neopixels live on pin 17
#define NUM_LEDS      10   // number of neopixels on the circuit playground
#define COLOR_ORDER GRB
 
// The right button will control the neopxiels' brightness.  Use these 
// varables to control how many brightness levels you'd like.
 
uint8_t brightness = 30;  // initial brightness level
uint8_t minbrightness = 10;  // minimum brightness level
uint8_t maxbrightness = 180;  //maximum brightness level -- can be set as high as 255
uint8_t brightnessint = 40;  //amount of brightness added for each button press.  
 
int STEPS = 10;  //makes the rainbow colors more or less spread out
int NUM_MODES = 5;  // change this number if you add or subtract modes
 
CRGB leds[NUM_LEDS];      // set up an LED array
 
CRGBPalette16 currentPalette;
TBlendType    currentBlending;
 
int ledMode = 0;       //Initial mode 
bool leftButtonPressed;
bool rightButtonPressed;
 
// SOUND REACTIVE SETUP --------------
// based on Adafruit's VU Meter code: https://learn.adafruit.com/led-ampli-tie/overview?view=all#the-code
 
#define MIC_PIN         A4                                    // Analog port for microphone
#define DC_OFFSET  0                                          // DC offset in mic signal - if unusure, leave 0
                                                              // I calculated this value by serialprintln lots of mic values
#define NOISE     200                                         // Noise/hum/interference in mic signal and increased value until it went quiet
#define SAMPLES   60                                          // Length of buffer for dynamic level adjustment
#define TOP (NUM_LEDS + 2)                                    // Allow dot to go slightly off scale
#define PEAK_FALL 10                                          // Rate of peak falling dot
 
byte
  peak      = 0,                                              // Used for falling dot
  dotCount  = 0,                                              // Frame counter for delaying dot-falling speed
  volCount  = 0;                                              // Frame counter for storing past volume data
int
  vol[SAMPLES],                                               // Collection of prior volume samples
  lvl       = 10,                                             // Current audio level, change this number to adjust sensitivity
  minLvlAvg = 0,                                              // For dynamic adjustment of graph low & high
  maxLvlAvg = 512;
 
 
// SETUP -----------------------------------------------------
void setup() {
  Serial.begin(57600);
  CircuitPlayground.begin();
  FastLED.addLeds<WS2812B, CP_PIN, COLOR_ORDER>(leds, 10).setCorrection( TypicalLEDStrip );
  currentBlending = LINEARBLEND;
  set_max_power_in_volts_and_milliamps(5, 500);               // FastLED 2.1 Power management set at 5V, 500mA
 
  
}
```

focalintent, 2017-09-17:
The pin definitions are different for every platform and MCU.  And because FastLED talks directly to the low level GPIO registers, for a board to be supported by FastLED a custom mapping of PIN numbers to low level GPIO registers needs to be made.  This has been done for the version of the circuit playground with an atmega32u4 — it hasn’t been done for a D21 based circuit playground.  This is why it isn’t working for you right now.

eduncan911, 2017-09-17:
Ah. Is this done within the FastLED library?

If so, could you point me to a little code for the atmega32u4?  I'll see
what I can do for a PR if I can figure it out.

patocarr, 2017-11-27:
I've managed to patch the d21/fastpin_arm_d21.h to add the pins for the Adafruit Circuit Playground Express. I tested 3 Arduino pins (6, 8 & 9) and they work fine. I had to #define the ADAFRUIT_CIRCUITPLAYGROUND_M0 in the sketch (as workaround) and set the code block before the ARDUINO_SAMD_ZERO block or it would take precedence. I'm not sure where this should be defined, or if it is already defined what value it should have. I'm not familiar enough with Arduino to figure this out.
You may find the current code in https://github.com/patocarr/FastLED/tree/Adafruit_CPE_pins 
If anyone knows how to place the pin definitions in the right spot with the proper #defines for CPE, I can make it a PR. @ladyada comes to mind, as she recently added Gemma & Trinket support. :)

ladyada, 2017-11-27:
fixed in latest commit plz try!

ladyada, 2018-01-11:
@kriegsman @focalintent hiya can you make a new release so this issue can be closed - thx! :)

kriegsman, 2018-01-12:
Thanks for the nudge; I'll talk with @focalintent this evening.

kriegsman, 2018-01-12:
I believe that release "3.1.7" should now be live with all the community contributions -- thank you so much for your ongoing support and contributions, here, and everywhere else, too.

closed 2018-01-12 by kriegsman

ladyada, 2018-01-12:
yaaaaay

kriegsman, 2018-01-13:
Aaaaand apparently I mean "3.1.8", because library.json.  Anyway, cheers, and thank you.

robboz4, 2018-07-23:
HI!

I'm trying to get an Adafruit Express running with this library. I get the pragma error message listed in the original post, but I have the latest library.  The code compiles and downloads but doesn't run. I tried several examples with the FastLED and they all do the same. I also saw this message in the compile output:
Using library FastLED-3.1.8 at version 3.1.6 in folder: /Users/daverobinson/Documents/Arduino/libraries/FastLED-3.1.8 

Could that be the problem? If so how do I fix it? 

The code is from Adafruit's Seashell project which used the classic board, but I have the express board whichI'd like to use.

ConstantinCreations, 2021-01-25:
Circuit Playground Express ATSAMD21 G18A-U 1843 still isn't compatible with the FastLED library.