FastLED/FastLED #503: RFDUINO Bluetooth will not work as i close the interrupt  #define FASTLED_ALLOW_INTERRUPTS 0

tdzuc, 2017-09-25:
i already read this post
https://github.com/FastLED/FastLED/wiki/Interrupt-problems
but i still don't know how to  use the similar method to deal with the bluetooth.
If i use  
#define FASTLED_ALLOW_INTERRUPTS 0
in rfduino for adafruit neopixel LEDs.
The bluetooth will not work properly.
It seems the bluetooth interrupts is closed, therefore the bluetooth will not work normally as i use FastLED.show() in 100ms.
 
Do you have any ideal?

focalintent, 2017-09-25:
Unfortunately the clock rate on the rfduino is low enough that you will have to choose between Bluetooth or shorter chains of neopixel LEDs.

closed 2017-09-25 by focalintent

tdzuc, 2017-09-25:
ok thanks if i change rfduino to simblee. Is it ok?

focalintent, 2017-09-26:
It looks like the simblee uses the same underlying nrf51822 mcu that the rfduino does - so you'd have the same problem there.