FastLED/FastLED #504: ESP32 + FastLED

PhilColbert, 2017-09-26:
Struggling to get this working ?

Have a setup with the 8266 working perfectly.

Using the same code and wiring setup, just the first 2 LED's light up white and the 3rd LED flickers.

Using 72 APA102 LED's and standard ESP32 dev board.

Wired directly to 5V, no level converter as it wasnt needed on the esp8266 board

Any ideas or anyone got this working yet ?

Thanks

focalintent, 2017-09-27:
@samguyer any thoughts for @PhilColbert?  I don't have any esp32 based setups here to test with, so I'm mostly relying on sam's pull request :)

samguyer, 2017-09-27:
@PhilColbert Which pin are you using? I have found that the ESP32 is very finicky about the pins.

PhilColbert, 2017-09-27:
Hi

Thanks for the reply, tried 13,14 and 25,26 so far....

Seemingly same result, first 2 leds on white, 3rd flashes....

Thanks

gnkarn, 2017-09-27:
You should try the level converter first , also in an ESP 8266  before anything else .

PhilColbert, 2017-09-27:
Yes, will do :)

thanks

focalintent, 2017-09-27:
Also, as a sanity check - make sure you haven’t gotten clock and data wired backwards (the behavior you’re describing is not dissimilar to what happens when the clock and data lines have been swapped)

PhilColbert, 2017-09-27:
:) Tried it both ways.... first thing to check that one, been there done that before several times, with them reversed, nothing comes out.

Thanks tho, will try Level converter soon first.

samguyer, 2017-09-27:
The code that I pushed for the ESP32 is mainly for clockless chips, like the WS2812 family. It was adapted directly from the ESP8266 code, with modifications mainly to handle interrupts. I would not expect to see problems with LEDs that have an explicit clock signal, but I'm pretty new to this stuff (and to hacking FastLED).

samguyer, 2017-09-27:
Also: the pins you're using should be fine, but there are so many weird exceptions on this processor. I have gotten things to work reliably on GPIO 18 and 19.

PhilColbert, 2017-09-27:
Ok, will give it a try , thanks.

PhilColbert, 2017-09-27:
Tried with level converter, works just the same on the 8266 with level converter.

With the esp32 and a level converter using pins 18 and 19, works exactly the same, first 2 led's are white and then third flickers.
( tried switching clock and data line to make sure its correct - no colour the other way )

Using FastLED version 3.001.006
"No hardware SPI pins defined.  All SPI access will default to bitbanged output"

Any idea where to start looking for the issue ?

Thanks

samguyer, 2017-09-27:
Do you have another strip, or a different kind of strip you can try?

samguyer, 2017-09-29:
OK, I think I've made some improvements to the ESP32 support. 
@focalintent: do you want me submit another pull request?

chrisparton1991, 2017-09-29:
I'm able to drive my UCS1903 strips with an ESP32 (Specifically the DoIt ESP32 Devkit V1 model) without the need for logic level conversion, using the latest FastLED pulled from master (3.001.006).

I have a different issue with FastLED (can't use 3-pin strips and WiFi at the same time, probably due to interrupts). I'll be raising a separate issue or asking in the community about that one though.

samguyer, 2017-09-29:
@chrisparton1991 I have been working on the interrupt problem on the ESP32, and I think I have a more stable implementation.

PhilColbert, 2017-09-29:
Tried a set of SK6812  - bit better - one wire version, they light up just not the correct ones , i tried led 0 and 20 and led 0 and 15 lit up ?

Do you have the updated code somewhere ? Thanks

samguyer, 2017-09-29:
You can try my branch here: https://github.com/samguyer/FastLED
All you need is the new copy of FastLED/platforms/esp/32/clockless_esp32.h

PhilColbert, 2017-09-29:
Thanks for the reply ! glad for the help.

Same , I have  strip of SK6812 - its nearly working ( I have set chipset to WS2812 as I think these are a clone  of ?

It nearly works - the correct leds light up with the commands, but not the right colour and now and then another one lights up.

Is it worth trying the level converter again - this did nothing last time on all the esp boards.

Thanks.

samguyer, 2017-09-29:
I don't think it's a voltage level issue. My guess is that interrupts are coming in at bad times and causing the showPixels function in FastLED to bail out early. What else do you have running on the ESP32 besides the LED code?

samguyer, 2017-09-29:
Here's something you could try: find the following comment in clockless_esp32.h in the showRGBInternal function, and comment out the conditional below it:

// if interrupts took longer than 45µs, punt on the current frame

I'd be interested to hear how that goes!

PhilColbert, 2017-09-29:
Nothing, just the LED code - couple of for loops to change the colours, thats it.

I am more interested in the APA102 pixels really, these are for my project, these are interrupt free arent they ?

Wheres best to start looking why these dont work ? :)

Thanks

samguyer, 2017-09-29:
Huh. That's a drag. It seems like the APAs (and any chip with a separate clock line) should work without any special support (there actually *isn't* any special support). You say it works for you on an ESP8266?

Debugging the clockless chips is beyond my abilities -- I think you really need an oscilloscope to do it right. Here is an excellent article about how the signal works. One thing to try might be to check and tweak the timing numbers. They are passed in as template arguments T1, T2, and T3 to the ClocklessController template.

https://wp.josh.com/2014/05/13/ws2812-neopixels-are-not-so-finicky-once-you-get-to-know-them/

One crazy idea that I got from this article is to enable interrupts during the transmission of 1 bits, since they can be arbitrarily long!

PhilColbert, 2017-09-29:
Code works perfectly on the 8266.

on the esp32 it just lights up the first 2 leds's white and flickers the 3rd whatever you send to it....

Just want to use the esp32 for the ble on board and the speed.... such a shame it doesnt work

I removed all the conditional code and same result .

Thaks

samguyer, 2017-09-29:
My only thought is that it has something to do with the specific pin and/or pin configuration that you're using. Here's the setup code I'm using successfully:

`  pinMode(18, OUTPUT);
  FastLED.addLeds<WS2812, 18, GRB>(g_LEDs, NUM_LEDS).setCorrection( TypicalLEDStrip );
  FastLED.setBrightness(g_Brightness);
  fill_solid(g_LEDs, NUM_LEDS, CRGB::Blue);
  FastLED.show();`

PhilColbert, 2017-09-29:
Tried that, all leds light up, but alternating red/blue/green along the strip.... 

thanks anyway :)

samguyer, 2017-09-29:
Oh, I forgot the call to pinMode. I assume you've got that in there, too.

focalintent, 2017-09-29:
1 bits can only be arbitrarily long for the first pixel - because of the signal reshaping that gets done, the hi gets dropped to low coming out of the first chip in the “expected” time and then you end up still running into the reset.

focalintent, 2017-09-29:
> Tried that, all leds light up, but alternating red/blue/green along the strip

Sounds like what I would expect to happen if you plugged an RGBW strip in and tried to drive it as an RGB strip

samguyer, 2017-09-29:
@focalintent OK, I didn't realize that the long 1-bit would not propagate to subsequent pixels. I'm not sure what the long-term solution is, but the code I have right now disables interrupts only for a very short time.

PhilColbert, 2017-09-29:
Hi, does sound like its a RGBW strip.... sorry guys, this is all new to me !

Does fastled have a chipset for RGBW ?

chrisparton1991, 2017-09-29:
@samguyer thanks so much for your hard work. Your changes fixed my hanging problem but I get strange output on my strips. Interestingly, the output is almost perfect when I drive the strips without any wifi code (some minor flickering, might just be my dodgy temporary connection).

Anyway, I know this is Phil's issue so I don't want to throw all of my problems on the pile :) I'll keep tinkering and see if I can figure anything out. In the worst-case scenario I'll just fork out for some APA102s.

samguyer, 2017-09-30:
@chrisparton1991 It sounds like an interrupt problem. I'll try to debug it with WiFi turned on.

samguyer, 2017-09-30:
@chrisparton1991 Can you send me your code? Or maybe just a simplified version for me to test, if you don't want to share it all.

chrisparton1991, 2017-09-30:
Yeah sure thing, I'm not home right now but I'll get it to you ASAP.
Appreciate the help :)

chrisparton1991, 2017-09-30:
@samguyer I've retested my strips and I'm getting issues even without any WiFi code in place (using your fork, which fixed my hanging issue). I'll need to do some more investigation. Interestingly, my various 5m LED strips perform differently despite all being UCS1903 from the same manufacturer.

Here's a gist of my code, for what it's worth. I don't think it'll be too helpful though, as it's about a simple example of FastLED as you can get.

https://gist.github.com/chrisparton1991/81228b569fa8238cd4deb209a7f67d41

I'll do some more testing to see if I can eliminate any possibilities. I'll let you know if I find anything.

PhilColbert, 2017-09-30:
I cant get the APA102's to work unfortunately, if anyone cant let me know !

Will try put the hack thats online to get the SK6812  RGBW strips working and make sure these are ok first :)

Thanks for all the help.

samguyer, 2017-09-30:
@chrisparton1991 That code works perfectly for me when I change it for WS2812 strips. Unfortunately, I don't have other kinds of strips to test.

chrisparton1991, 2017-09-30:
No problem @samguyer, thanks for looking into it. I've decided to use ESP8266 chips for the time being. I'll definitely test APA102s on my ESP32 when I get around to purchasing them.

Redferne, 2017-10-01:
Gave the 3.1.6 release a spin on a ESP32 with a 300 LED WS2813 strip and could simply not get it to work at all. I've adjusted the timings a bit to get T1H within the specs. Tried a few settings but found these to be closest to what the spec says. The strip is dark and all LEDs will come on in a rainbow fashion when I reset reset the ESP. I'm running the demoreel. Got a 5V 20A supply hooked up this this thingy and the voltage is OK. Haven't tried with external HCT drivers (3.3V on the datapins), however on previous projects WS2813 worked fine with 3.3V. I guess I need to find a ESP8266 somewhere in the mess to compare.

`class WS2813Controller : public ClocklessController<DATA_PIN, NS(320), NS(500), NS(940), RGB_ORDER> {};`

I hooked up the Hantek and checked the timing:

T0H ~370ns  Spec says: (300-450ns)
[image]

T1H ~850ns Spec says: (750-1000ns)
[image]

T0L and T1L ~945ns Spec says: (300ns-100µs)
[image]

focalintent, 2017-10-01:
Yeah - i'm not going to take timing changes - the esp32 clockless code probably needs to be fixed/tweaked so that it outputs in line with the values set in the template definitions (until I get back to working on the library - that ball is in @samguyer 's court :)

tbnobody, 2017-10-01:
I just made several experiments adding the RMT approach for controlling the ws2812 instead of the bit banging one.

For the ws2812 (I currently don't have other stripes available) it works very well.
You can see my changed file here https://github.com/tbnobody/FastLED/blob/master/platforms/esp/32/clockless_esp32.h

In my opinion, the usage of static variables and methods is not yet so nice but required to implement the interrupt handling (does someone now another way to avoid the static methods?)

The other thing where I am unsure is whether I interprete the timing values T1, T2 and T3 correctly. It works for ws2812 but as mentioned I cannot test it currently with other chipsets.

focalintent, 2017-10-01:
The math is, roughly:

```
T0H == T1 
T0L == T2 + T3
T1H == T1 + T2
T1L == T3
```

On the esp32, does RMT work for any pin (it looks like the code is setting the pin, but unsure how flexible it actually ends up being)

tbnobody, 2017-10-02:
Based on the datasheet: https://espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf RMT should be available on any pin.

Redferne, 2017-10-02:
@tbnobody

Great stuff. Will have a spin of your code tomorrow and check the timing. Thanks!

samguyer, 2017-10-02:
@tbnobody That is cool. I had been planning to try the same thing, but had not gotten around to it. One question: does your current implementation limit the total number of pixels?

tbnobody, 2017-10-02:
@samguyer currently I don't see a limit in the pixel amount beside the available memory. I compiled my source for testing purposes with a count of 600 LEDs and everything is working as expected (I currently have only 60 LED's physically available but there wasn't any problem)

tbnobody, 2017-10-02:
Hm one limiation is maybe the amount of parallel pins on which the output occurs. Currently the RMT channel 0 is used for all instances of the ClocklessController class. And as far as I understood it, FastLED supports the parallel output to different pins with different pixel types.

samguyer, 2017-10-02:
@tbnobody Awesome.

samguyer, 2017-10-02:
I've discovered a little more information about using an ESP32 to drive clockless strips. Caveat: right now, I only have WS2812 strips.

The showRGBInternal function has a conditional that makes sure that if interrupts are enabled they don't take too much time to service. The conditional looks like this:

```
           // if interrupts took longer than 45µs, punt on the current frame                                                                             
            if((int32_t)(__clock_cycles()-last_mark) > 0) {
                if((int32_t)(__clock_cycles()-last_mark) > (T1+T2+T3+((WAIT_TIME-INTERRUPT_THRESHOLD)*CLKS_PER_US))) {                                                                                                                            
                    return 0;
                }

From what I can tell, this situation is happening a lot. That is, interrupts *often* take too long to keep the bits going fast enough. I tried removing the conditional while still allowing interrupts, and it causes the LEDs to flicker a lot.

Turning off interrupts for the entire call to showRGBInternal (i.e., but setting FASTLED_ALLOW_INTERRUPTS to 0) fixes the problem completely, but I'm not sure what kind of side-effects that might have on other services, such as WiFi.

My thinking is that the RMT approach might be the right way to go, since the timing of the bit-banging is handled entirely by the RMT subsystem.

tbnobody, 2017-10-03:
I just made a rough implementation using a different RMT channel for each assigned pin: https://github.com/tbnobody/FastLED/blob/master/platforms/esp/32/clockless_esp32.h

The following code mirrors the stripe on two different ports:
```
FastLED.addLeds<NEOPIXEL, 18>(leds, NUM_LEDS);
FastLED.addLeds<NEOPIXEL, 19>(leds, NUM_LEDS);
```
Due to the fact that only 8 RMT channels are available, the amount of ClocklessController instances is limited to 8. In my opinion, thats maybe a acceptable restriction.

samguyer, 2017-10-04:
@tbnobody Very cool.

One question about the behavior: will FastLED.show() return before the RMT has finished sending the bits? Is that the purpose of allocating a separate buffer? (i.e., to avoid a race condition)

tbnobody, 2017-10-04:
@samguyer this part of the code is still from the original authors. But due to the fact that the memory is freed at the end of the function (and also looking at the semaphore code), I would assume that everything is sent when the function exists. (if not, there is a "use after free" bug). 

But I know what you are getting at. I will try to get rid of this additional buffer :)

samguyer, 2017-10-04:
I'm happy to help -- I was planning on working on this same idea myself. I did find one other implementation that looked a little different. It might be worth checking out:

https://github.com/Lucas-Bruder/ESP32_LED_STRIP

samguyer, 2017-10-04:
@tbnobody Looking at it more carefully, I think you're right: the semaphore causes showPixels() to wait until the data is all sent.

samguyer, 2017-10-05:
@tbnobody Starting with Lucas Bruder's code, I realized that we can use a much simpler method based on the RMT function rmt_write_items. It allows an arbitrary length signal to be sent without explicit use of interrupts. Check out my first cut here:

https://github.com/samguyer/FastLED/blob/master/platforms/esp/32/clockless_esp32.h

It's not fully ported yet -- the timing numbers are hard-wired for WS2812, and I don't have your feature of incrementing the RMT channel for multiple instances. BUT the code is so much simpler.

tbnobody, 2017-10-05:
@samguyer you are right. This code is much easier. But as far as I see it, it's allocating pixels.size() * 3 * 8 uint32_t variables (rmt_item32_t) at the beginning of the function. 
With 60 LED's that are (60 * 3 * 8 * 32 = 46080bit / 8) = 5760 byte.

The interrupt method which I commited a few minutes ago (without the additional buffer required) is just using the RMTMEM variable which consumes 64 * 8 uint32_t = 2048byte.

With more LED's it even gets worse. I am not sure whether it's possible to reduce the memory consumption with Lucas Bruder's code by calling rmt_write_items multiple times with a smaller amount of memory. (but how to measure the time between each rmt_write_items call?)

samguyer, 2017-10-05:
@tbnobody That's a fair point, although the ESP32 has quite a lot of memory. Maybe I'll try calling rmt_write_items multiple times with a limited size buffer. That should be equivalent to the strategy based on interrupts.

Have you thought about how to avoid calling malloc/free every time the strip is rendered? I don't have a good solution to that problem.

tbnobody, 2017-10-05:
You can avoid the malloc/free call by using a static variable (e.g.) and fill it with a portion of the FastLED data. But then you are just one step away from the interrupt method. It's also using a "static" variable called RMTMEM, which is already defined in the espressif header files. 

The problem with calling rmt_write_items multiple time is maybe the time BETWEEN each call. Because I would assume it's not possible to keep the right timing between each call. The interrupt method is updating the first half of the send buffer while the second half is beeing processed by the RMT subsystem. That leads to the exact timing. I think that is only possible with the non-interrupt approach by sending the whole stripe at once.

samguyer, 2017-10-06:
@tbnobody It looks like you're right: I was not able to structure the signal as a series of calls to rmt_write_items, even if I let them run asynchronously while I compute the next chunk. I tried a lot of crazy things, and I did get the time between each call low enough (theoretically), but it still didn't work. I'm not sure why. It looks like the interrupt driven approach is the only one that works with a limited size buffer.

My only reservation about Chris Osborn's code is that there is so much stuff in there that I don't understand. Unless you object, I might take a pass over that code and see if I can simplify it. The rmt_write_items function in the ESP32 core basically does the same thing (fills half the buffer at a time and waits for an interrupt to fill the next half), but doesn't seem to use nearly as many special configuration calls and flags.

tbnobody, 2017-10-06:
@samguyer no no,I have no objects that you simplify my code. Simple and short code is always good :)

[175 more replies]