FastLED/FastLED #506: ESP32 hangs with FastLED, 3-pin strip and UDP

chrisparton1991, 2017-09-29:
I'm using an ESP32 with FastLED and a UCS1903 LED strip, which receives data over UDP. One packet of data is successfully processed, then the code hangs on the next `udp.beginPacket()` call. As soon as I comment out my `FastLED.show()` call, the hanging no longer occurs.

I can change my code to use any 4-pin strip type (e.g. APA102) and it runs fine (without powering any strips, since I don't have any 4-pins to test). All of the 3-pin strip types have the same issue.

My understanding is that FastLED disables interrupts for 3-pin strips due to their strict timing requirements. I've previously had Wifi+UDP running on an ESP8266 with a 3-pin strip, so I'm not sure if this is a systemic issue or just a bug.

Here's my code stripped down to more or less the bare minimum. https://gist.github.com/chrisparton1991/28eaf638e668409f9d2742008d090a57

I have a DoIt ESP32 Devkit V1 board and I'm using FastLED v3.001.006.

chrisparton1991, 2017-10-04:
No longer an issue with the various ESP32 experiments that have been going in #504

closed 2017-10-04 by chrisparton1991

glararan, 2017-10-22:
I have similar problem, can you share how did u exactly fix it?

chrisparton1991, 2017-10-22:
Hi @glararan, take a look at #504, there's some good work being done to overcome this issue. As far as I'm aware, the current solution is to take https://github.com/samguyer/FastLED/blob/master/platforms/esp/32/clockless_esp32.h and use that in your local FastLED installation until it gets merged in (assuming it does).

glararan, 2017-10-22:
okay thanks man! I will edit post if I get it working tomorrow

zombodotcom, 2018-02-02:
How did you fix this? I added clockless_esp32.h

Do I add a header for it or something?