FastLED/FastLED #508: Will not compile: Receiving "Arduino/libraries/FastLED/pixelset.h:298: error: 'template<class PIXEL_TYPE> class CPixelView' used without template parameters "

DangerouslyExplosive, 2017-09-30:
I am using an older version of the Arduino IDE, 1.0.6, and the library is preventing me from compiling. Should I try to use another version of the IDE (1.8.1) or is there a way for me to fix the problem myself? I am using a Mac running OS X 10.5.8, and my options as far as resources are limited. I can't download the newer versions of the IDE, because they don't support my OS anymore, and I have to download my libraries on a different computer or find them on a site other than GitHub because my browser's plugins don't support the GitHub downloader. Ideas, anyone?

marmilicious, 2017-09-30:
If you want to try the current FastLED version I just downloaded it to my google drive and dropbox so maybe one of those will work for you.
https://drive.google.com/file/d/0B1DBHDKW1M8ubGNrVzFXQVVnYWs/view?usp=sharing
https://www.dropbox.com/s/zj0edzbe38tdzsp/FastLED-master%203.1.6%20%289-30-2017%29.zip?dl=0

I believe I was using Arduino 1.8.1 for a while just fine (and now 1.8.3), so yes, upgrade to that version if you can.

focalintent, 2017-09-30:
1.0.6 hasn't been supported for a while - and the library has had a number of updates that probably rely on C++ features from newer versions of GCC than what 1.0.6 has.

closed 2017-09-30 by focalintent

ScottieD369, 2018-01-03:
This doesn't explain how to solve this issue?
I want to make it work with older versions. 
It used to work for me like months to years ago......

jasoncoon, 2018-01-03:
If you want to use an old version of Arduino, you will likely need to find and use an old version of FastLED as well.

focalintent, 2018-01-03:
There is no solving this issue - there are portions of FastLED that now rely on more recent versions of GCC.