FastLED/FastLED #509: Some colors too dark for hsv2rgb_rainbow on computer screen

Jochen0x90h, 2017-10-03:
I tried out to use hsv2rgb_rainbow on an ordinary computer screen and it looks that these ranges are too dark:
orange to yellow
aqua to blue
purple to pink

When I load the FastLED "rainbow" hue chart into Gimp and use the pipette on yellow (on left side of dark line), red is 229 and green is 215. According to the rgb graphs (and the code), the output of hsv2rgb_rainbow is 170 for red and green.
When using the pipette between purple and pink, I get 177 for red and 190 for blue (and 80 for green), but sould be 128 for red and blue and 0 for green.

Does anybody know where the difference comes from? Do I miss some postprocessing of the rgb values before displaying them?

kriegsman, 2017-10-03:
The on-screen chart was postprocessed for visual clarity, so it’s not merely the numeric output of the function. 
Are you seeing problems with the function output itself? (Basically: ignore the screen chart, and just look at the function output in the code.)

Jochen0x90h, 2017-10-03:
The only problem for me is that my rainbow doesn't look as nice as the one in the wiki. yellow looks muddy and pink too dark. But I think the code is correct numerically. So i could start to tune it to make yellow and pink more bright unless you have another idea.

focalintent, 2017-10-03:
Also - don't forget, the library is tuned for leds - which throw off light differently than what you see on a monitor screen.  Don't tweak your color code too much for the screen, it'll look differently weird on your leds later...

kriegsman, 2017-10-03:
As @focalintent says, the color space that you get on a screen is fairly different from a color space that you'd get on LEDs.  I've found that if you make colors "look right" on the screen, the same color numbers come out desaturated, pastel, and pale on LEDs.  If you make things "look right" on the LEDs, they come out dim and, as you said, muddy, on the screen.  

I'm going to close this ticket since I think it's not actually a library bug, but I do recommend that you continue the conversation on the FastLED discussion group on G+, http://fastled.io/+

closed 2017-10-03 by kriegsman