FastLED/FastLED #511: It would be nice to be able to calibrate the power management

naelolaiz, 2017-10-14:
I'm testing the library and so far it works great, I love it. Thank you!
But reaching the power management functions I measured that the real current does not correspond to what I asked for (the diff was of 100mA). Would it be possible to add some way to enter empiric measures of the used device in order to calibrate the power management?

marmilicious, 2017-10-17:
Could you use a copy of
https://github.com/FastLED/FastLED/blob/master/power_mgt.cpp
for your project and edit lines 25-28 to values closer to what you're reading?

marmilicious, 2017-10-17:
You're right though, it would be nice to be able to define something at the top of your code for this.