FastLED/FastLED #513: Define one in two leds as the actual strip

Colu1245, 2017-10-28:
Hello everybody,

I wanted to have a matrix with leds spaced by 6 cm from each other. 
To reduce  soldering time, I simply bought WS2812b strip (1 led every 3cm).
However, I only need to illuminate one led every 6 cm. Therefore, one in two leds is useless.

Is it possible to modify the FastLED library somewhere to take into account this perpetual offset ?
I would like to use for example "fill_solid" and only illuminate the usefull leds.

Thank you for help !

marmilicious, 2017-10-28:
Github is used for reporting bugs in the library.  The FastLED G+ group might be a better place to ask this and get more ideas.  Here's one idea that doesn't involve modifying the library that might work for you.

Create two CRGB arrarys, one to work with and one that gets displayed.
CRGB working[NUM_LEDS/2];
CRGB leds[NUM_LEDS];

In your main loop run a function before calling show() that spaces out the pixels.

void loop() {
  fill_solid(working, NUM_LEDS/2, CRGB::Red);

  SpaceOutPixels();  //run before calling FastLED.show()
  FastLED.show();
}

//function to space data out to every other pixel
SpaceOutPixels() {
  for (uint16_t i=1; i<NUM_LEDS/2; i++) {
    leds[i] = working[2(i-1)];
  }
}

Colu1245, 2017-10-29:
Thanks for your reply.
This seems to be the best way to do it. I will try.
And sorry for having posting this thread here.

closed 2017-10-29 by Colu1245