FastLED/FastLED #515: GradientPalette behaving unexpected

maxxh11, 2017-10-30:
I tried using a Gradient Palette and get strange results. I'm not sure if im using it incorrectly or  if there's a bug inside the Library. 

Following [this](https://github.com/FastLED/FastLED/wiki/Gradient-color-palettes) wiki page I'm doing the following steps: 
1.  'DEFINE_GRADIENT_PALETTE ( heatmap_gp )'
2. Create a 'CRGBPalette16 gPal'
3. Point the gPal to heatmap_gp 
4. Use `leds[0] = ColorFromPalette( gPal, 255); ` to assign a color from the gradient. 
5. `FastLED.show();`

In my understanding the led should turn on and be pure white with a Brightness of 255. But in reality I get a dim greenish color.

The issue goes away if I use CRGBPalette256 in step 2 instead of CRGBPalette16. 

Am I doing something wrong or is this a bug ?

marmilicious, 2017-10-31:
In order to support smooth blending for "circular" palettes, each color blends with "the next", and in the case of the last color, "the next" is the first color since it wraps around, so the last 15 numbers are blending back toward index 0.  Sometime circular blending is very useful, other times it might not be what you want.  You can use index 240 as your last value to avoid blending.

If you would still like to code using a 0-255 range but have the output maximum be 240 and avoid wrapping you can use the scale8 function:
ColorFromPalette( gPal, scale8( i , 240 ) );

closed 2019-01-20 by focalintent

systembolaget, 2021-12-09:
[Here](https://github.com/marmilicious/FastLED_examples/blob/master/palette_example2.ino) it says the last stop must be 255 though.

marmilicious, 2021-12-10:
Hello @systembolaget, yes, the last index position defined in the gradient palette must be 255.
See line 1662 here:  
http://fastled.io/docs/3.1/colorutils_8h_source.html

> //  IMPORTANT NOTES & CAVEATS:
> //
> //  - The last 'index' position MUST BE 255!  Failure to end with
> //    index 255 will result in program hangs or crashes.

systembolaget, 2021-12-10:
Thanks for confirming that.

Still, it is unclear how the stops do work over the range of 256, see my problem [here](https://www.reddit.com/r/FastLED/comments/rcs16k/apa102c_70_led_strip_palette_stops_and_colour/?utm_source=share&utm_medium=web2x&context=3) if you have a minute.

CorpusCallosum, 2022-03-27:
From my understanding, you need to use 240 as the last color stop to prevent the circular blending. Repeat the last color twice, using stops 240 and 255, but don't go above 240 for the index.

For example:
`DEFINE_GRADIENT_PALETTE( test ) {
  0,     0,  0,  0,  //black
64,   255,  0,  0,  //red
128,   0,255,  0,   //green
240,   0,0,255,     //blue
255,   0,0,255 };   //blue again (this one gets ignored)`

marmilicious, 2022-03-27:
More discussion here:  
https://www.reddit.com/r/FastLED/comments/fx6w94/colorfrompalette_incorrectly_mapping_across_0240/

For all general discussions, help, and how-to, please use the FastLED Reddit group here: 
https://www.reddit.com/r/FastLED