FastLED/FastLED #516: SPI Slave Select pin ???

Serveurperso, 2017-10-31:
Documentation is pretty bad about this, where I can setup the slave select pin ?

https://github.com/FastLED/FastLED/wiki/SPI-Hardware-or-Bit-banging

focalintent, 2017-10-31:
Fastled doesn’t use space select at all, since none of the supported led chipsets use it.  (Supporting doing that is part of a planned rewrite of the spi parts at some point, for folks who have wired up their own slave select hardware).

Serveurperso, 2017-10-31:
I can't use my simple SPI multiplexer ? bad:(

Serveurperso, 2017-10-31:
please support the basic 74HC595 chip !

focalintent, 2017-10-31:
Supporting the *595’s requires driving the pwm in software with interrupts which is something I took out of the library years ago when I started supporting a mix of platforms. 

As far as slave select support - you can do it yourself by pulling your choice of slave select pin down low before calling FastLED.show().

Serveurperso, 2017-10-31:
Thanks, you can close. Because I switch to colorduino a better hardware solution than shift register to drive the 8x8 matrix

closed 2019-01-20 by focalintent