FastLED/FastLED #518: Can FastLED be modified for PL9823 LEDS?

JohnMustang442, 2017-11-06:
I've made a 5x5x5 led cube using PL9823 leds.  Using the Examples in the FastLED library, the leds beyond #70 randomly flicker different brightness, beyond #100 flicker incorrect colors as well.  The power supplied is solid 5V at 10A common grounded to Arduino.  Same results with 16MHz Uno or Nano.
I re-arranged the data injection to a different point on the cube.    i.e. injected the data at led #60 and wrapped the output of #125 back to #1, and find the problem moved down the chain to the new #70 led and beyond.

Looking at the signal SENT by FastLED to the first PL9823, I see timing of:
T0H = 250NS     
T0L = 1000NS        
T1H = 900NS
T1L = 400NS

The OUTPUT of my first PL9823 is:
T0H = 400NS
T0L  = 850NS
T1H = 800NS
T10 = 450NS
These values are out of spec according to the PL9823 datasheet.
The datasheet for PL9823 shows:
T0H = 350NS      +-150NS
T0L = 1360NS       +-150NS
T1H = 1360NS       +-150NS
T1L +  350NS       +-150NS

Could this timing error cause my problem with leds past #70 in the chain?
Is there something I can adjust in FastLED to correct this timing error?

I found an animation code example written by someone named Josh, that WORKS well. It lights all 125 leds with no flicker or incorrect colors.
The timing it causes to the input of the first PL9823 is:
T0H = 400NS
T0L = 1300NS
T1H = 900NS
T1L  = 1250NS

Thanks,  John

focalintent, 2017-11-06:
What version of arduino are you using?  The avr code is clock cycle counted asm - and the definitions are in line with the data sheet but not what you’re seeing - what you’re seeing makes it seem like gcc is doing some optimization, cutting out cycles.  

Alternatively - the timings you are reporting look more like ws2811 timings - are you sure you set the led chipset correctly in the code?

JohnMustang442, 2017-11-06:
I have two boards I've tried. One says UNO with the MEGA328p chip, has 12MHz crystal near the CH340 chip and a 16MHz crystal near the MEGA328p chip.  The other board says NANO3 with the same Mega328p on it.  I tried to attach is a picture of them here.  Also the PL9823 datasheet...
How do I tell what version they are?  Is there a correction that I enter into my code dependent on which version I learn I have?  Here is the code I'm testing.

#include <FastLED.h>
#define NUM_LEDS 125
#define LED_PIN 1
#define FASTLED_INTERRUPT_RETRY_COUNT 1
CRGB led[NUM_LEDS];
void setup() {
  // put your setup code here, to run once:
FastLED.addLeds<NEOPIXEL,LED_PIN>(led, NUM_LEDS);
for (int i = 0; i <NUM_LEDS; i++) {
     led[i] = CRGB(15,20,3);
}
FastLED.show();
}
void loop() {
  // put your main code here, to run repeatedly:
for (int i = 0; i <NUM_LEDS; i++) {
     led[i] = CRGB(2,2,2);
     //if i = 20, set pin 2 = 1
    }
FastLED.show();
delay(20);
}

JohnMustang442, 2017-11-06:
Sorry, I tried to upload (drag and drop) the pl9823 datasheet and a picture of my boards, but I guess my browser fails me.  Or I did it wrong.

JohnMustang442, 2017-11-06:
How do I set the led chipset in the code?

JohnMustang442, 2017-11-06:
I added the define_type pl9823 to my code add still get random brightness after ~led 70. Also the same timing results as I listed in my first post.
CODE:
#include <FastLED.h>
#define NUM_LEDS 125
#define LED_PIN 1
#define LED_TYPE PL9823
CRGB led[NUM_LEDS];
void setup() {
  // put your setup code here, to run once:
FastLED.addLeds<NEOPIXEL,LED_PIN>(led, NUM_LEDS);
for (int i = 0; i <NUM_LEDS; i++) {
     led[i] = CRGB(15,20,3);
}
FastLED.show();
}
void loop() {
  // put your main code here, to run repeatedly:
for (int i = 0; i <NUM_LEDS; i++) {
     led[i] = CRGB(20,20,20);
    }
FastLED.show();
delay(20);
}

focalintent, 2017-11-06:
Do you see this line?

```
FastLED.addLeds<NEOPIXEL,LED_PIN>(led, NUM_LEDS);
```

That's telling FastLED to configure the less as NEOPIXEL's which are WS2812B leds - you need to change that line to be:

```
FastLED.addLeds<PL9823,LED_PIN>(led, NUM_LEDS);
```

Then it will use the PL9823 timings instead of the WS2812B timings.

JohnMustang442, 2017-11-06:
Thank You very much. That works great, has perfect timing and works my cube great!
I started to do that once but noticed it didn't turn red like the NEOPIXEL entry so I didn't follow through with the attempt.  I'm new to the arduino. Sorry for the petty bother. My bad!

closed 2017-11-06 by JohnMustang442