FastLED/FastLED #519: WS2812B doesn't keep its color; sometimes flickering

Agent-3P1C, 2017-11-10:
Hello,
I have got a problem with my new WS2812B Strip.
The strip is at three points connected to a 5V power supply. A Arduino nano is connected to the beginning of the strip with a digital pin. 
When I upload some example sketches to the arduino nano it may react in two different ways. 
Either it is always flickering white. However I can see a little bit the colors it should have on some LEDs. But they are also flickering white. It seems like the white is "overwriting" the colors. 
Or the LEDs show, for example the one correct color, however after a few seconds big chunks appear white or in some different color. So the color it should be is not staying. 
What can I do to have the colors I wish displayed?
Thank you very much

focalintent, 2017-11-10:
Do you have a connection between ground on the strip and ground on the arduino nano?

Agent-3P1C, 2017-11-10:
I didn't want to connect them because the "ground" on the strip is (-). And the arduino gets its power through the usb port.
Do i still have to connect them?

focalintent, 2017-11-10:
Yes - it's important for there to be a common ground between the arduino and the leds - see https://learn.adafruit.com/adafruit-neopixel-uberguide/basic-connections

Agent-3P1C, 2017-11-10:
Oh my ^^
Thanky you
works like a charm
Sorry for that inconvenience

closed 2017-11-10 by Agent-3P1C