FastLED/FastLED #524: FastLED compile error in Particle cloud

mcflie1994, 2017-11-18:
Hello,

Thank you for your wonderful library and all the work that is put into it. I have been attempting to use the library with the Particle Cloud IDE, but when I go to compile my (admittedly quite simple) test code, I receive an error. The error message is 

> lib/FastLED/led_sysdefs.h:19:43: platforms/avr/led_sysdefs_avr.h: No such file or directory

The full raw error is: 

`Processing  led_strip_test.ino
Checking library FastLED...
Installing library FastLED 3.1.5 to lib/FastLED ...
Library FastLED 3.1.5 installed.
make -C ../newlib_nano 
make[1]: Entering directory '/firmware/newlib_nano'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/firmware/newlib_nano'
make -C ../user 
make[1]: Entering directory '/firmware/user'
Building cpp file: src/led_strip_test.cpp
Invoking: ARM GCC CPP Compiler
mkdir -p ../build/target/user/platform-31src/
arm-unknown-linux-gnueabi-gcc -DPLATFORM_THREADING=0 -DPLATFORM_ID=31 -DPLATFORM_NAME=raspberrypi -DUSBD_VID_SPARK=0x1D50 -DUSBD_PID_DFU=0x607F -DUSBD_PID_CDC=0x607D -DEMBEDDED_TARGET=0 -g3 -O3 -gdwarf-2 -Wno-unused-local-typedefs -DINCLUDE_PLATFORM=1 -DPRODUCT_ID=31 -DPRODUCT_FIRMWARE_VERSION=65535 -DSYSTEM_VERSION_STRING=0.6.0-rc.1 -DRELEASE_BUILD -I./inc -I../wiring/inc -I../system/inc -I../services/inc -I../communication/src -I../hal/inc -I../hal/shared -Isrc -I./libraries -Ilib/FastLED/src -I. -MD -MP -MF ../build/target/user/platform-31src/led_strip_test.o.d -ffunction-sections -fdata-sections -Wall -Wno-switch -Wno-error=deprecated-declarations -fmessage-length=0 -fno-strict-aliasing -DSPARK=1 -DPARTICLE=1 -DSTART_DFU_FLASHER_SERIAL_SPEED=14400 -DSTART_YMODEM_FLASHER_SERIAL_SPEED=28800 -DSPARK_PLATFORM_NET=gcc -fno-builtin-malloc -fno-builtin-free -fno-builtin-realloc  -DLOG_INCLUDE_SOURCE_INFO -DMODULE_VERSION=0 -DMODULE_FUNCTION=3 -DMODULE_DEPENDENCY=0,0,0 -D_GNU_SOURCE -D_WINSOCK_H -DLOG_MODULE_CATEGORY="\"app\""  -fno-rtti -fcheck-new -std=gnu++11 -c -o ../build/target/user/platform-31src/led_strip_test.o src/led_strip_test.cpp
In file included from led_strip_test.ino:2:0:
lib/FastLED/src/FastLED.h:12:2: warning: #warning FastLED version 3001000 (Not really a warning, just telling you here.) [-Wcpp]
 #warning FastLED version 3001000  (Not really a warning, just telling you here.)
  ^
In file included from lib/FastLED/src/FastLED.h:34:0,
                 from led_strip_test.ino:2:
lib/FastLED/src/led_sysdefs.h:19:43: fatal error: platforms/avr/led_sysdefs_avr.h: No such file or directory
 #include "platforms/avr/led_sysdefs_avr.h"
                                           ^
compilation terminated.
../build/module.mk:264: recipe for target '../build/target/user/platform-31src/led_strip_test.o' failed
make[1]: *** [../build/target/user/platform-31src/led_strip_test.o] Error 1
make[1]: Leaving directory '/firmware/user'
../build/recurse.mk:11: recipe for target 'user' failed
make: *** [user] Error 2`

The Library is pulled from the particle library manager. The version is 3.1.5 and is listed as published by Daniel Garcia and Mark Kriegsman. If there is something I'm doing wrong, please let me know. Thank you. :)

~
Martin

[image]

focalintent, 2017-11-18:
What board are you trying to build for?

mcflie1994, 2017-11-18:
Particle pi. Its a raspberry pi model b rev 2 (I believe, I know its not rev 1, 2 USB ports, full size SD and mounting holes). current firmware on the device is v0.5.1.

I chose particle pi due to the fact that I want to ethernet instead of wifi and I have 3 of them around.

focalintent, 2017-11-18:
Fastled doesn’t support the pi, that’s why you’re getting errors :)

mcflie1994, 2017-11-18:
Darn. Okay, I'll see if I can find a different ethernet supported board. Thanks.

closed 2017-11-18 by mcflie1994

mcflie1994, 2017-11-18:
I would like to ask though, if there may ever be support for the pi as they have particle support, even though they aren't an MCU platform. Thanks

focalintent, 2017-11-18:
Unlikely to be direct support - fastled likes talking to hardware at a level of control that’s annoying to do on Linux. Some folks have talked about doing a fork of FastLED for Linux though - I don’t know if anyone has done it though.

mikeburg, 2017-11-18:
Well, there is the pull request for a FastLED linux port . . . https://github.com/FastLED/FastLED/pull/422

The drawback is only 4-pin leds are supported.

focalintent, 2017-11-18:
And if you see my comments on there - I'm unlikely to pull that back into FastLED proper - I don't want to support a linux port :)