FastLED/FastLED #525: Can't compile on ESP 32 (__cxa_pure_virtual)

axwell, 2017-11-19:
Hello,
can't compile any example on ESP32.
IDE: AtomIO

Error:

c:/users/afdin/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp
32-elf/lib\libstdc++.a(pure.o):pure.cc:(.text.__cxa_pure_virtual+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\featheresp32\firmware.elf] Error 1

Any ideas ?

focalintent, 2017-11-21:
@samguyer any thoughts?  (Note: technically the arduino IDE is the only one that's tested/supported for building FastLED)

samguyer, 2017-11-21:
No idea. I've only used the Arduino IDE.
@axwell , can you compile other ESP32 applications that don't use FastLED?

samguyer, 2017-11-21:
There is a discussion on Stack Overflow about this error:

https://stackoverflow.com/questions/920500/what-is-the-purpose-of-cxa-pure-virtual

It looks like FastLED defines the __cxa_pure_virtual function (in FastLED.cpp), which then conflicts with the one in the standard library.

@focalintent Is there a way to add some conditional directive around these definitions? I'm just not sure what the right test would be.

focalintent, 2017-11-21:
They’re already guarded by an ifdef - something like NEED_CXX_BITS or Some such.

axwell, 2017-11-21:
@samguyer I can compile other applications.
This pull request fixed the issue: https://github.com/FastLED/FastLED/pull/495
After this pull request LEDS are not responding, and this pull request fixed issue: https://github.com/samguyer/FastLED/blob/master/platforms/esp/32/clockless_esp32.h

Now everithing is working perfectly.

samguyer, 2017-11-21:
@axwell Great! Happy blinking!

closed 2019-01-20 by focalintent