FastLED/FastLED #526: .show() blocking loop?

the-magister, 2017-11-21:
In .show():
void CFastLED::show(uint8_t scale) {
	// guard against showing too rapidly
	**while(m_nMinMicros && ((micros()-lastshow) < m_nMinMicros));**
...

That looks like a blocking while() loop.  Would it introduce problems to change this to use the yield() function?  For an AVR, this probably translates to a NOP.  For devices with other "background" tasks (I'm looking at you, ESP8266), this would allow the WiFi stack to get serviced.

void CFastLED::show(uint8_t scale) {
	// guard against showing too rapidly
	**while(m_nMinMicros && ((micros()-lastshow) < m_nMinMicros)) { yield(); }**

focalintent, 2017-11-21:
It is blocking, but only for cases where a maximum frame rate has been specified (e.g. for the 400hz specified by WS2812 leds, that would mean it'd block for a maximum of 2.5ms) - given how long show often blocks for without calling yield while writing out led data (milliseconds, if not dozens of milliseconds), I'm not sure that a yield here would buy you much.

focalintent, 2017-11-21:
In fact, if i remember right, ESP8266 wifi is handled via interrupt, and interrupts will be handled in this while loop (they don't get disabled until writing out led data).

the-magister, 2017-11-21:
When doing 20-33 fps animations set by the library's (nice) frame rate
settings, I note that I have more WiFi problems if I let the library handle
the frame rate.  When I handle it outside the library (using delay and
yield) I get less.

Yes, WiFi will fire an interrupt, but the yield() gives over time to the
handling "other, mysterious stuff that I don't fully understand" including
something vaguely described as "the wifi stack":

> *Yielding*This is one of the most critical differences between the
> ESP8266 and a more classical Arduino microcontroller. The ESP8266 runs a
> lot of utility functions in the background – keeping WiFi connected,
> managing the TCP/IP stack, and performing other duties. Blocking these
> functions from running can cause the ESP8266 to crash and reset itself. To
> avoid these mysterious resets, avoid long, blocking loops in your sketch.

I'm wondering if this change would get us out of answering approximately
1.5 million questions a week around "my esper-thinger keeps crashing" and
the like?  That is, this change and considering setting the library default
frame rate to something reasonable might save newcomers a lot of hassle?

Anyway, tendered for consideration.  Great library, and thank you.

focalintent, 2017-11-21:
Don't use setMaxRefreshRate for this!  setMaxRefreshRate should really only be used for cases like the WS2812 where, if you update them at a higher rate than 400hz they start to freak out.  

And no - this yield wouldn't change the esp crashing problems (that has to do with interrupts being blocked, not yield not being called) and the default max refresh rate is uncapped (and set automatically to 400hz when using leds like the WS2812 that glitch visibly when you update them at higher than 400hz).  I think you're the first person I've heard of that's attempting to use setMaxRefreshRate to control animation speed.

The proper way to manage _frame rate_ (note that this is different from refresh rate) is by using delay  (either delay or FastLED.delay) and/or the EVERY_N_MILLISECONDS macros and such.

the-magister, 2017-11-21:
Effectively, I'm using something like EVERY_N.  More accurately, a timer
function tracking the time since last .show().

And, apologies, but the setRefresh business, on the face of it, does appear
to a natural way to control frame rate.  Given the tight loop, someone
(like me) could definitely trip the ESP8266's wdt by using it.  I may or
may not be speaking from experience.  ;)

closed 2018-06-23 by focalintent