FastLED/FastLED #527: Getting hue

gubagu, 2017-11-21:
Hi there,
Amazing library ! I came across am interesting case that i cannot solve myself.

Is there a way to get the hue or the color from a running led ?

Thank you in advance!

Cheerz, Guby

KilianB, 2017-11-27:
Isn't the color saved in the "color array" you have specified at the beginning? The color value of the leds can be found here:

`CRGB leds[NUM_LEDS];`

Regarding color space and the data type you can find all the information you need in this documentation: https://github.com/FastLED/FastLED/wiki/Pixel-reference.

If you need the exact values after color and temperature correction it's worth a shot to see how **computeCorrection** in [controler.h](https://github.com/FastLED/FastLED/blob/64d729e6e6d6bbb8da239e65963d6eb82a7059a7/controller.h)  is used.

gubagu, 2017-11-28:
Thanks! I'll see how far i can get with this info 😃

closed 2017-11-28 by gubagu