FastLED/FastLED #531: It looks like FastLED isn't compatible with STM32F103C microcontroller

tutankhamen, 2017-11-29:
It's weird because I can see

#elif defined(STM32F10X_MD)

in platform.h, but still getting:

fatal error: avr/io.h: No such file or directory

stevstrong, 2017-12-07:
You should use in your code
#elif defined (ARDUINO_ARCH_STM32F1)

ajkeeton, 2017-12-30:
I tried that and now FastLED is searching for application.h which isn't included anywhere I've seen

nk64, 2018-01-02:
I was able to get my Blue Pill (Generic STM32F103C Series) working with FastLED by following the instructions in this comment by @evq  https://github.com/FastLED/FastLED/issues/142#issuecomment-78129532

The main point for me was to manually merge in https://github.com/evq/FastLED/tree/stm32f103
which at the time boiled down to these 5 files:
"FastLED\platforms\arm\stm32\led_sysdefs_arm_stm32.h"
"FastLED\platforms\arm\stm32\clockless_arm_stm32.h"
"FastLED\platforms\arm\stm32\fastpin_arm_stm32.h"
"FastLED\led_sysdefs.h"
"FastLED\platforms.h"

I also did the code change to reduce the sketch size dramatically:
https://github.com/FastLED/FastLED/issues/142#issuecomment-219462643

ajkeeton, 2018-01-20:
Any interest in trying to get a pull request open? If not I might fork and try to get the ball rolling

evq, 2018-01-20:
@ajkeeton Last time I offered to PR it didn't seem like there was interest. Maybe that's changed now that there's a larger community around the Blue Pill?

Please feel free to fork and attempt a PR, I'd love to see it merged and maintained!

I would suggest cherry-picking the following commit, there's a missing pin definition. 
https://github.com/treeoftenere/FastLED/commit/8bea7143bdd3735ffcf66a665144346291592a24

There's also this branch https://github.com/treeoftenere/FastLED/commits/tenere with parallel output. <3 @shlomozippel

eosrei, 2018-01-25:
:+1: I'm using @evq 's branch with PlatformIO and a maple mini clone: https://github.com/eosrei/FastLED-STM32/commits/stm32f103

kolyshkin, 2018-02-15:
https://github.com/FastLED/FastLED/pull/568

closed 2019-01-20 by focalintent

AndKe, 2019-02-19:
Why is this issue closed ?
I got 
```
/home/andre/Arduino/projects/libraries/FastLED/platforms/arm/stm32/led_sysdefs_arm_stm32.h:4:25: fatal error: application.h: No such file or directory
 #include <application.h>
                         ^
compilation terminated.
```

with blue pill & FastLED 3.2.6

painthink, 2020-12-23:
avr/io.h: No such file or directory

Hackswell, 2020-12-23:
You need to pick the proper board.  It's probably Generic Black Pill STM32F103 or some such.  Make sure you pull the newest FastLED because they included a lot of new stuff in a new release a few days ago.

AndKe, 2020-12-23:
Happily, I just updated to 3.4.0  :)
Then I got:
```
home/andre/Arduino/projects/libraries/FastLED/src/led_sysdefs.h:53:2: error: #error "This platform isn't recognized by FastLED... yet.  See comments in FastLED/led_sysdefs.h for options."
   53 | #error "This platform isn't recognized by FastLED... yet.  See comments in FastLED/led_sysdefs.h for options."
```
In led_sysdefs.h I see
```
#elif defined(STM32F10X_MD) || defined(__STM32F1__)
```
Yet I cannot compile with CoreSTM32's "Generic 32F103C6"  "BluePill F103C8"  or "BlackPill F103C8"

The Chrismas grinch must have done that :(

AndKe, 2020-12-28:
@focalintent   - please reopen this issue.

shlomozippel, 2020-12-29:
> @focalintent - please reopen this issue.

Sadly, he will not be reopening any more issues :(

As a bandaid - try adding `|| defined(STM32F103xE)` or whatever appears in the `product_line` for the board you're using in CoreSTM32 boards.txt (https://github.com/stm32duino/Arduino_Core_STM32/blob/master/boards.txt)

kriegsman, 2020-12-29:
Would it be useful to open a fresh issue for this? Or should we continue it here?

AndKe, 2020-12-30:
The bluepills I use have this define:  : GenF1.menu.pnum.BLUEPILL_F103C6.build.product_line=STM32F103xB
So I I changed led_sysdefs.h 
#elif defined(STM32F10X_MD) || defined(__STM32F1__) 
to
#elif defined(STM32F10X_MD) || defined(__STM32F1__) || defined(STM32F103xB)

this gave me another error:
FastLED/src/platforms/arm/stm32/led_sysdefs_arm_stm32.h:24:2: error: #error "Platform not supported"
   24 | #error "Platform not supported"

so I changed in that file:
#elif defined (__STM32F1__)
to 
#elif defined (__STM32F1__) || defined(STM32F103xB)

- finally, I changed all "if defined" to accept STMF103xB   like this:

```
$ grep -inr "__STM32F1__"
src/led_sysdefs.h:27:#elif defined(STM32F10X_MD) || defined(__STM32F1__) || defined(STM32F103xB)
src/platforms/arm/stm32/led_sysdefs_arm_stm32.h:16:#elif defined (__STM32F1__) || defined(STM32F103xB)
src/platforms/arm/stm32/fastpin_arm_stm32.h:63:#elif defined(__STM32F1__)  || defined(STM32F103xB)
src/platforms/arm/stm32/fastpin_arm_stm32.h:127:#if defined(__STM32F1__) || defined(STM32F103xB)
src/platforms/arm/stm32/fastpin_arm_stm32.h:170:#endif // __STM32F1__
src/platforms.h:27:#elif defined(STM32F10X_MD) || defined(__STM32F1__) || defined(STM32F103xB)
andre@loke:~/Arduino/projects/libraries/FastLED$ 

```
which only gave me this horror-show after initially promising almost-compilation:
https://gist.github.com/AndKe/60b7764042407f0b7dcc55986cc916d7