FastLED/FastLED #536: attiny88 woes

moosepr, 2017-12-06:
Howdy

I have come across your library while trying t get round some issues I am having while trying to make an attiny88 work with a WS2812 LED

I had tried the adafruit neopixel library to no avail, so i thought i would give fastled a go

I am using attinycore (https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/extras/ATtiny_x8.md) and arduino and am unable to get any of your examples to work. The led is attached to the pin PA3/PCINT27 which links to pin 26 in arduino terms, however fastled has issues with that pin number

`/home/petay/sketchbook/libraries/FastLED/fastpin.h:207:2:` error: static assertion failed: Invalid pin specified
  static_assert(validpin(), "Invalid pin specified");
  ^
exit status 1`

although after further testing, it seems to have issue with an pin!!

focalintent, 2017-12-07:
Can you turn on verbose compile output and dump the full output into gist.github.com and link to it here? 

Right now, FastLED has no pin/port definitions for the ATTiny 88 -- those are all defined in platforms/avr/fastpin_avr.h (these need to be defined because FastLED, at compile time, turns pin accesses into direct io register accesses for performance - but this means that the pin number to io register/port mapping has to be setup and known at compile time, not runtime.

moosepr, 2017-12-07:
sure, the dump is here

https://gist.github.com/moosepr/264d71722ea4d52d93001210e045c68d

i did expect it may be the case of nobody being mad enough to try it on an attiny88 before :P

brickstuff, 2018-11-03:
Hi, did this ever get fixed? I have an ATTiny88 as well and can’t get it to work with FastLED. I’ve looked at the fastpin-avr.h file and see all the chip defs but don’t know enough about the structure of how this is written to add a new section for the 88. It’s a nice chip, with hardware SPI and I2C which is almost unheard of with Tinys, so having FastLED support would be amazing.

moosepr, 2018-11-03:
I tried the adafruit library instead but that wouldn't work either! In the end I moved to an atmega 328 and that solved all my issues

brickstuff, 2018-11-03:
Yeah, that's not an option in my case-- pricing for the 328 is too high for my application.  I'll keep plugging away with the 88.

moosepr, 2018-11-03:
I got it working on an 85, but software i2c was a little patchy, and I'm assuming that you are in need of the extra pins

brickstuff, 2018-11-03:
Correct, not just the pins but hardware SPI support.  The 88 is a really neat little chip in that way.

moosepr, 2018-11-03:
Yeah it is a great chip. I'm not sure if it's the lower clock speed, or some kind of interrupt issue

brickstuff, 2018-11-03:
I'm going to see if I can add pin definitions on my own.  Need to make a large pot of coffee and spend a few hours with datasheets, but it has to be possible.

moosepr, 2018-11-03:
Good luck!

leachim6, 2019-04-11:
Was this ever fixed in any future iterations of the library?
I'm having the same issues with Attiny88.

brickstuff, 2019-04-11:
> Was this ever fixed in any future iterations of the library?
> I'm having the same issues with Attiny88.

As far as I know, nothing has been done.  I was unable to get it to work, which is too bad, because I have another project coming up where the 88 would be a great fit.