FastLED/FastLED #537: Makerfocus D1 Mini NodeMcu (ESP8266) + WS2811 daisy chain flicker

Drivium, 2017-12-08:
Following tutorial here: https://www.youtube.com/watch?v=6Y6jUM1OaYM and purchased everything from the description.  All set up and working beautifully...except for flickering down the string of LED strings.  First string has no flickering, but gets progressively worse with each daisy chained string.

Here is my wiring setup: https://imgur.com/a/jZaoc

Here is all of the hardware I'm using:
LED Lights: Rextin 50PCS WS2811 IC Pixels Digital Addressable LED String Lights Waterproof RGB Full Color 12mm DC 5V (50PCS) 
Raspberry PI: CanaKit Raspberry Pi 3 Complete Starter Kit - 32 GB Edition 
Arduino: Makerfocus D1 Mini NodeMcu 4M Bytes Lua WIFI Development Board Base on ESP8266 ESP-12F N Compatible NodeMcu Arduino 
Power Supply: ALITOVE 5V 60A 300W Power Supply Transformer Adapter Converter AC110V/220V to DC 5V 60amp Charger for WS2812B WS2811 WS2801 APA102 LED Strip Pixel Light

Things I've tried:
1. Adding these lines: #define FASTLED_ALLOW_INTERRUPTS 0 and #define FASTLED_INTERRUPT_RETRY_COUNT 0 (described here: https://github.com/FastLED/FastLED/issues/306)

2. Rearranging #Includes as described here: https://forum.pjrc.com/threads/24934-Confounding-Flicker-with-WS2812-strips-and-OctoWS2811-on-Teesny-3-1

No success so far.  Admittedly, this is the first time I've ever worked with ANY of these components.  

This is the code in the description of the tutorial I'm using:
https://raw.githubusercontent.com/Snipercaine/Holiday-LED-files/master/OutsideLEDs_PublicOTA2.ino

Any help would be very much appreciated.  I'm waiting to hang them until I figure this out and we're nearly midway through Dec!  :)  Thanks in advance.

Blink515, 2017-12-08:
>    threads/24934-Confounding-Flicker-with-WS2812-strips-
>    and-OctoWS2811-on-Teesny-3-1
> files/master/OutsideLEDs_PublicOTA2.ino

Add the 2 lines of code as explained here
<https://youtu.be/s5yLLtPrKAM?t=4m50s> and it should resolve the issue.

Drivium, 2017-12-08:
Not sure how I missed this video.  Thank you!  I'll give this a go and report back.

Blink515, 2017-12-08:
You are injecting power every so often down the strip of LEDs too right?
I'm not sure if Dr Zzs mentions having to do that in his videos or not.
Just having a large power supply isn't sufficient since the voltage will
drop after long runs of wire.

Drivium, 2017-12-08:
@Blink515 I haven't yet, but plan to.  I did verify that injecting power down the line DOES increase brightness, but has no affect on flickering.

Drivium, 2017-12-10:
Made all of the mods from the video, but flicker persists.  Not sure what else to try.

the-magister, 2017-12-10:
Your pi is 3.3v. The strips take their signals at 5v. See "level shifter".
Make sure the level shifter can handle the signal rate.

---
Mike (@mobile)

MFornander, 2017-12-10:
Yes you need a level shifter.  This is what we used to get everything working with hundreds of LEDs adn 8266: https://www.tindie.com/products/jasoncoon/wemos-d1-mini-esp8266-level-shifter-mini-shield/

thigger, 2017-12-11:
Another thing to consider - are they definitely WS2811 and not 2812B? The timing requirements are slightly different and I've just fixed a flickering issue using ESP8266:

https://plus.google.com/102754903922110930885/posts/UfBwA5eiNbL

Drivium, 2017-12-11:
@thigger yea, according to their packaging - WS2811.  @the-magister @MFornander  Level shifter ordered.  I'll report back once received, installed, and tested.  :)

focalintent, 2017-12-12:
I’ve started seeing reports of leds marked WS2811 but wanting WS2812 timings - so also try setting the led type to WS2812 or WS2812B

Drivium, 2017-12-12:
@focalintent Ok, I'll give that a shot as well.

brendanmatkin, 2018-01-05:
This might be old but I've also had that happen once with a bad LED in the chain (they were flickering red only). Try re-arranging your strips and see if it helps? 

In my case the flickering started way past the actual bad LED (i.e. the flickering wasn't apparent until at least a few dozen LEDs past the bad one).

skullthug, 2018-07-26:
@Drivium what ever became of this issue? Did the level shifter do anything for you?
I ask because I'm in the near exact same setup & situation.

closed 2019-01-20 by focalintent