FastLED/FastLED #538: ATMega 1284 not working out of the box (50kHz instead of 800kHz)

neomonkey, 2017-12-11:
Hello,

I have problems with my new ATMega1284P chip. I used logic analyzer to check the pattern which is ok but the timing is 50 kHz.

The setup:
- I use the newest Arduino IDE with newest FastLED with MCUdude's MightyCore https://github.com/MCUdude/MightyCore
    o I use 16MHz
    o I use BOD 2.7V
    o I use standard pinout (MightyCore)
    o LTO Disabled
    o Chip MightyCore 1284P <- with the P

- I rip the hex and program it with RPI3 (Programming successful)

- I use AVRdude to program the chip with the following tutorial (changed the chip to atmega1284p and checked that everything is ok. Programming successful) https://learn.adafruit.com/program-an-avr-or-arduino-using-raspberry-pi-gpio-pins/programming

The code:
- I use FastLED demo : "'Moving' a LED"
- I use pin 3 (AIN1 and PWM)
- LED SK6812Mini (I use WS2812B in the LED choosing) with GRB setup. The timing was almost the same in both specs.
- delay 30

Result:
Pattern is correct but the logic analyzer tells me that the speed is 50kHz.

Is there something which I do wrong or is there a bug? I can run tests to get this working. I have no idea how the FastLED library works so I hope you could provide me with some help with this one.

focalintent, 2017-12-12:
What you're seeing in the logic analyzer is making me think that your chip is running at 1Mhz, not 16Mhz (50kHz is 1/16th the 800kHz that ws28xx and clones want).

Can you also do a build with verbose output and paste the build log into a gist.github.com and link to it here?   I want to see what compile time defines MightyCore is setting (since I think this is the first reference i've seen to someone using MightyCore w/FastLED).

focalintent, 2017-12-12:
>>> Select your microcontroller in the boards menu, then select the clock frequency. You'll have to hit "Burn bootloader" in order to set the correct fuses and upload the correct bootloader.  

Just verifying that you did this as well.

neomonkey, 2017-12-12:
<script src="https://gist.github.com/neomonkey/285effd6cd8b771d607d4561f9584576.js">ATMEGA 1284P verbose with wrong fuses</script>

Okay, I though that the fuses go automatically (new with AVR stuff). It seemed that it didn't do that. 

So it seems the problem is there. Then I went to MCUdude/MightyCore board file and ripped the fuse values and tried to learn what they mean (quite simple actually). I did only one change (mistake?) and removed JTAG.

I uploaded the fuses with AVRdude command with slower baud rate (I took some random tutorial. I don't know why this was done with slower but I wanted to follow the tutorial). Everything else were correct I triple checked how I sent it (no erase)

`-U efuse:w: .. :m -U ...`

Ok, so the fuse values were (in the same order):

efuse: 0xFD
lfuse: 0xF7
hfuse: 0xD6

The I checked that okokokok. Then I tried to reupload the hex (with bootloader) and it was not able to connect ICSP anymore. So I think I kinda bricked it..

I have bought spare but I don't want to brick them all. Do you happen to know did I made the mistake on taking the JTAG off?

Well now I understand where the problem is (now I understand more about the AVR). But could you throw me some help with the fuses?

AND

Well it seems that you guys don't have a proper test environment with the ATMega1284P so I can also run those if necessary! I uploaded the AVR verbose. Hope that helps.

focalintent, 2017-12-12:
>I have bought spare but I don't want to brick them all. Do you happen to know did I made the mistake on taking the JTAG off?
> Well now I understand where the problem is (now I understand more about the AVR). But could you throw me some help with the fuses?

I don't - i've done no work at all with bare AVR chips - mostly I work with supporting the various arduino boards and clones (as well as a handful of arm and other 32-bit based platforms)

> Well it seems that you guys don't have a proper test environment with the ATMega1284P so I can also run those if necessary!

Given the number of platforms that I support, and the relatively low number of users that are using the 1284P (I think you're maybe the 3rd or 4th I've run across in the last 7 years), it's not a realistic platform for me to put a lot of direct effort into.  I'm happy to fix bug reports when they come up if they're easy enough to do without requiring major code base changes/testing.

neomonkey, 2018-01-22:
Hi.

Okay now I got this working. This issue might be a tutorial for other people who want to play with ATMEGA1284P and FastLED. Here goes:

1. You need to buy external 16MHz crystal and you need 2x 22pF capacitors.
2. Make wiring AREF1 -> Crystal1 -> 22pF_1 -> GND and AREF2 -> Crystal2 -> 22pF_" -> GND
3. Setup the RPI with https://learn.adafruit.com/program-an-avr-or-arduino-using-raspberry-pi-gpio-pins/overview to make the RPI AVR programmer

4. I use the newest Arduino IDE with newest FastLED with MCUdude's MightyCore https://github.com/MCUdude/MightyCore
o I use 16MHz
o I use BOD 2.7V
o I use standard pinout (MightyCore)
o LTO Disabled
o Chip MightyCore 1284P <- with the P

5. Setup the fuses to 
efuse: 0xFD
lfuse: 0xF7
hfuse: 0xD6

CMD: sudo avrdude -p atmega1284p -C ~/avrdude_gpio.conf -c pi_1 -v -U efuse:w:0xFD:m -U lfuse:w:0xF7:m -U hfuse:w:0xD6:m

closed 2018-06-23 by focalintent