FastLED/FastLED #540: attiny84a crashing

xxv, 2017-12-12:
I'm having some trouble with getting APA102Cs to work with some attiny84as. I've managed to create a minimal failing test case here:

```
#include <FastLED.h>

const static uint8_t NUM_LEDS       = 1;

CRGB leds[NUM_LEDS];

APA102Controller<6, 4, BGR>ledController;

void setup() {
  FastLED.addLeds((CLEDController*) &ledController, leds, NUM_LEDS);

  leds[0] = CRGB(20, 20, 0);
  FastLED.show();
  delay(500);
  leds[0] = CRGB(20, 20, 20);
  FastLED.show();
  delay(500);
}

void loop() {}
```

This compiles just fine, but causes non-deterministic behavior when run (presumably crashing of some sort). The LED changes to a bunch of different colors not specified in the script. If the CRGB value is changed to `CRGB(20, 0, 0)` it behaves properly.

Some details about the setup: I'm using PlatformIO to compile. Using FastLED at commit c1ab8fa (it failed for 3.1.6 too). gcc version 5.4.0. and Arduino 1.10620.1 (provided by PlatformIO)

focalintent, 2017-12-12:
Please try building/installing using a recent version of arduino directly - platformIO isn’t fully supported for building,  also make sure that the port/pin mapping that is being used is sensical (the behavior you describe is sometimes seen if the clock/data puns are backwards).  

I believe there are pin definitions in fastpin_avr.h for the attiny84 - I don’t know if they would get picked up when building for the attiny84a (if they’re no, you’d see a warning/message pragma about pins not being defined in build output)

xxv, 2017-12-13:
I compiled/deployed the above sketch using FastLED 3.1.6, Arduino 1.8.5, and version 1.0.2 of the attiny core from my Debian machine. Unfortunately, while the hex output seems to be different, the result is the same. 

I tried reversing clock/data, and the results were worse (and it still crashed). I'll try to look at it with my logic analyzer and see what I can find there, but I'm not very hopeful there.

Any suggestions for a debugging approach?

focalintent, 2017-12-13:
The attiny is kind of a shit platform to test on (And why the trinket is the only officially tested/supported tiny :) — can you turn on verbose compiler output and paste a build log into gist.github.com so I can see what defines you are getting.  

Also - why are you declaring the controller statically yourself vs the standard way of calling addLeds?

xxv, 2017-12-13:
Hey now, just because they're small... ;-) (I like them because they're cheap and I'm already designing my own boards anyhow)

https://gist.github.com/xxv/ecefb813fd02866a80befdda5b51777e

I had sub-classed the controller to add support for APA102C global dimming. I'm not using that subclass for this, though.

I just tried:

```
#include <FastLED.h>

const static uint8_t NUM_LEDS       = 1;

CRGB leds[NUM_LEDS];

void setup() {
  FastLED.addLeds<APA102, 6, 4, BGR>(leds, NUM_LEDS);

  leds[0] = CRGB(20, 20, 0);
  FastLED.show();
  delay(500);
  leds[0] = CRGB(20, 20, 20);
  FastLED.show();
  delay(500);
}

void loop() {}
```

And am getting the same results, unfortunately.

xxv, 2017-12-13:
I'm still looking through scope output. The clock signal seems fine (though I'm not sure why my SPI decoder is getting so confused. I'm still pretty new to using Logic, so perhaps I don't have it configured properly): http://xxv.so/0xdec252f7

I'm seeing a +/- 0.15vDC 20khz fluctuation on the power line to the micro, but I don't imagine that's too problematic (I'm not an EE).

focalintent, 2017-12-13:
And you’re sure they’re apa102 and not sk6822?  The protocols are slightly different (but the sk6822 are sold as apa102 clones) - does it never write out the second frame 500ms later? 

Also - for laughs, try setting NUM_LEDS to something like 24 or 32 - it’s possible there’s a bug with framing when a small number of leds are used. 

Also -

xxv, 2017-12-13:
I got them off AliExpress, so it's possible they're different, but I have been successfully driving this batch of them from ESP8266es using FastLED 3.1.6 without any trouble.

I just tried with NUM_LEDS set to 24 with different, but still broken behavior.

The most confusing part is that it behaves differently each time it powers on. There's clearly some sort of crash/corruption going on, because it's toggling one of the other pins on my board that's attached to an LED (which isn't even set up as output in this sketch). 

I tried putting a 0.1uF decoupling cap on the power line, but that doesn't seem to improve things. Perhaps I need to move it physically closer to the micro, but I doubt it.

focalintent, 2017-12-13:
Have you verified that the pin/port mappings that are defined in fastpin_avr.h (see https://github.com/FastLED/FastLED/blob/master/platforms/avr/fastpin_avr.h#L83-L93 ) line up with the pin numbers you are using and the actual port/pins being used? 

Outside of the pin/port definitions, there's no difference between this and the other supported avr platforms - and this problem hasn't popped up anywhere else that's been reported :)

Also, what happens if you move your led writing code to loop instead of setup?

xxv, 2017-12-13:
Yeah, I verified that the pin/port mappings were right. This was confirmed by my scope.

I'll give the loop code a shot soon, but I'm going to guess that it won't change much.

So far my best suspicions are:

* Some dumb hardware design error (I am relatively new to digital electronics)
* Something involving interrupts
* An OOM error somehow
* Some bit of assembly that works on other AVRs, but has quirks on the attinys

xxv, 2017-12-17:
After screwing around with this for more time than I care to admit, I discovered that it's likely an issue with the older attiny Arduino cores. I was able to make this work on an attiny841 and an attiny84 using [ATTinyCore](https://github.com/SpenceKonde/ATTinyCore) instead of the tiny/tinymodern libraries. In doing this, I was able to get it going using the attiny841's hardware SPI driver. I suspect it may even be possible to use ATTinyCore's USI-emulated SPI driver on the attiny84 too, so I'll give that a shot when I get a chance. As I can't seem to reproduce this bug anymore after updating libraries, I'll close this.

closed 2017-12-17 by xxv

xxv, 2017-12-19:
Just for those who may stumble across this ticket: I've been digging into this more and found out that adding a 100nF bypass/decoupling capacitor across VCC/GND of the microcontroller fixes the reset problem I was having. I see other designs that use 10uF bypass caps too, but don't know nearly enough about this sort of thing to have any opinions on what the correct value cap to use is.