FastLED/FastLED #542: FASTLED not working with ESP8266 Node MCU

samiraa78, 2017-12-20:
Hi!

I am trying to use ESP8266 to  connect and control my LED from Hassio, the code i am using uses FASTLED however for some reason none of the demos of FAST LED or the code work, sometimes the first light lits and sometimes the light led lits thats all happens

Adifruit and all other library work fine any idea if i need to enter specific code for Node MCU

I am using WS2812B LED lights

xkuyax, 2017-12-20:
Have you got a level shifter? The nodemcu has only a logic level of 3.3 Volts and the leds have a minimum logic level of 3.5V, it might work (it works for me), but it might not work for every led strip/controller combination.

coryking, 2017-12-22:
Have you tried different pins?

tonton81, 2017-12-22:
use pins 5 or 14
i did a demo using pin 14 of this github source
Tony

tonton81, 2017-12-22:
and no level shifter either, 144 leds strip runs very well!

dwc56, 2017-12-25:
did you get it working because i am trying do do the same thing?

samiraa78, 2017-12-25:
Yes i dot it working in the code you need to change the pin address if you are mentioning D4 it wont work however if you only 4 it will work try doing that after couple of days of head banging this simple trick worked for me

Muplex, 2017-12-31:
I've got the same issue. Neopixel works with adafruit library but not with fastled. How did you solve it?

samiraa78, 2017-12-31:
What is the exact Pin number you have mentioned in the code if the pin is D4 and you have mentioned D4 trying changing it to 4 and see if it works

Muplex, 2018-01-02:
indeed it works!!!! Thank you very much!!!!

I want to explain the situation for future readers;
In my sketch I was using buttons + RGB Neopixel 16 dots ring.

The confusion comes out because with FastLed.H different object initialization (in these case the neopixel) in witch I had to use the number print on the nodemcu (D0 = 0; D1 = 1; ...... FOR FASTLED RELATED OBJ) BUT to initialize the buttons I had to map the  pins as known (D0 = 16; D1 = 5; .......... FOR THE BUTTONS)...

brendanmatkin, 2018-01-05:
As mentioned above NodeMCU re-maps the ESP8266 pins so they all line up nicely on the breakout. This image makes it pretty clear which is which (so if you want GPIO4 you can use D2, or GPIO3 is D9 (shown as RX on this picture)).

[image]

closed 2018-06-23 by focalintent

ArcadeNut, 2018-12-23:
I have the same issue.  I tested the adafruit Neopixel and it works with my WS2812B strip.  I'm using a Wemos D1 R1 board.  The strip turns on but erratic flashing.  i've use D5 or 15 and D4 or 4 no luck.  Its not a power issue as I'm power with a 5v 60a power supply and only 60LEDs.  I have a 470ohm resistor on the data line.

TheOnlyJoey, 2019-06-09:
> I have the same issue. I tested the adafruit Neopixel and it works with my WS2812B strip. I'm using a Wemos D1 R1 board. The strip turns on but erratic flashing. i've use D5 or 15 and D4 or 4 no [...]

Sorry if i am bumping a way to old issue.
If you are feeding the neopixel with a 5v line, your trigger value will be 3.5v, which is 0.2v to high for the gpio levels of your board (with a max of 3.3v).

shawheenattar, 2019-06-13:
Also sorry for bumping this but I'm also experiencing a similar issue. I got a level shifter to try to fix the problem, but it still isn't working with FastLED, although it does work with NeoPixel library without the level shifter. I've attached a pic of my wiring, so does anything look wrong? Would really appreciate the help. I've been banging my head on this for a few days now.

[image]

shawheenattar, 2019-06-13:
So looks like this was a bug with FastLED that's been fixed in v3.2.9. Everything works now after updating from v3.2.8 and without using the level shifter

ramanraja, 2019-12-10:
https://github.com/FastLED/FastLED/wiki/ESP8266-notes