FastLED/FastLED #544: There doesn't seem to be a way to set an entire strip to one color, without making an array to represent each LED.

SephReed, 2017-12-29:
I've been diving through the code looking for a way to not use an array for color setting (as I don't have the ram space left), and I can't seem to find one.

As far as I can tell,  CFastLED::addLeds creates a CLEDController which later has CLEDController::show() called, which then calls showPixels() which is a virtual function that is deabstracted by each different type of strip protocol class, and always requires an LED[] array.

The issue here is that I can not find a way to circumvent the array aspect.

What I would like to do is set 150 leds on a strip to the same color.  I can't afford the 450 bytes (~25% of an Arduino Uno's Ram space) to make a CRGB array.  I totally can afford to send 3 bytes of data down the entire line, though.

Am I missing something here?  How does one set an entire strip to a single color without making tons of redundant data?  And if it can't be done, how can I modify FastLED to make it capable of doing it?

focalintent, 2017-12-29:
There is, on the individual controller objects, a showColor method - that you can give an arbitrary number of leds to show a color for.  For example, you could do:

/// Tell the first controller (aka what was setup with the first call to addLeds) to show the color 
/// 255,127,130 on the first 1024 leds off that pin with a brightness of 255.
FastLED[0].showColor(CRGB(255,127,130), 1024, 255);

Then you can lie in addLeds about how many leds are on the strip/pin.  

Of course, at this point then, why are you bothering with using addressable leds?  It would be better to just have straight up non addressable rgb led strips driven by a pwm controller for the entire strip.

SephReed, 2018-01-04:
>It would be better to just have straight up non addressable rgb led strip

Working with what I've got.  Addressable were left over, low budget project.

Anyways, I'll try it out.  Thank you very much.

closed 2018-01-04 by focalintent