FastLED/FastLED #545: WS2813 Teensy3.1 unusual behavior beyond ~60 leds

VoswijkC, 2017-12-29:
Recently got a large string of WS2813 to control with a Teensy3.1

I loaded up a very simple rainbow test, but the strip behaved rather unusual and i have no possible explanation as to how.
-The entire string and code works perfectly fine with a normal arduino so the strip,Code and power supply aren't faulty.
-A small 24led string connected to the Teensy work well aswell. So my level shifter seems to be working normal too.
-Also works pretty well if the program is set to assume the strip is pretty small.
-when it is asked to add more than ~65 leds it goes haywire with flickering and whatnot.

Apparantly once i ask the library to add more than a certain amount of leds (around 60) on the Teensy it stops working properly with the set amount of leds turning off and the rest showing random colours. Perhaps there is a bug in the Fastled code itself?

This is with fastled release 3.1.6

focalintent, 2017-12-29:
Try disabling interrupts and see if that helps.  See https://github.com/FastLED/FastLED/wiki/Interrupt-problems - the ws2813 changed some of the timings around reset, and the code hasn’t been fully updated to account for that (and probably won’t be for a while, still).

VoswijkC, 2017-12-29:
Ah. that seems to do the trick! Did not know fastled allowed interrupts to be handled between each led on ARM devices.

Thank you very much for the assistance!

closed 2018-06-23 by focalintent