FastLED/FastLED #546: FastLED with SAMD21E17

alejandroImagina, 2018-01-08:
Hi, I've been using the FastLED library with an Atmega 2560 for a long time without any problem. But in my new project I have to use it with a SAMD21E17 and I couldn't get it to work :(

I'm now working with a MattairTech MT-D21E. I connect the Leds (WS2812B) to the PA02 (pin 2 in the Arduino IDE), and also defined "ADAFRUIT_TRINKET_M0" before the #include "FastLED.h", because it use the SAMD21E18 (the same CPU with double SRAM and Flash).

The Arduino IDE compile and upload the code correctly (no error or warnings), but the Leds didn't turn on.

I make a function to turn it Red, Blue or Green, and every time it's been called it writes a debug in the SerialUSB so I'm sure it is called.

This is the function, the defines and the Setup:

`#include <Wire.h> //
#define ADAFRUIT_TRINKET_M0
#include "FastLED.h"
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>

#define OLED_RESET 3
Adafruit_SSD1306 display(OLED_RESET);

#define NUM_LEDS 3
#define DATA_PIN 2

CRGB leds[NUM_LEDS];

void setup() 
{
  //Iniciamos los canales serie
  SerialUSB.begin(115200); //
  Serial1.begin(115200);  // SERCOM0
  Serial2.begin(115200);  //SERCOM2
  Serial3.begin(115200); //SERCOM3
 Wire.begin(); //SERCOM1

  //Setup Leds
  FastLED.addLeds<WS2812B, DATA_PIN, RGB>(leds, NUM_LEDS);

  //LCD I2C
  display.begin(SSD1306_SWITCHCAPVCC, 0x3C); //LCD I2C
  display.display();
}

void loop()
{
 ... //More Code but irrelevant to the Leds
 if(SerialUSB.available())
 {
   serialRead[0]=SerialUSB.read();
   switch(SerialRead[0])
   {
    case 'R':
       SerialUSB.println("Turning Red");
       leds[0]=CRGB::Red;
       leds[1]=CRGB::Red;
       leds[2]=CRGB::Red;
       FastLED.show();
       break;
   }
 }
 ...//More Code but irrelevant to the Leds
}
`

I've also check the PIN with a Logic analyzer and I saw nothing, every time I called the function I get the "Turning Red" through the USB but nothing more.

Maybe the problem is that I've all the SERCOM used in other tasks?

Thanks for the help.

focalintent, 2018-01-08:
I’m not sure it is enough to put the define where you have it - since that won’t ensure that it’s set for the handful of C++ source files the library has. 

Also you should check and make sure that you have the PIN number to port mapping right - those are defined explicitly in FastLED.   Also, I don’t know offhand if the trinket define will also get you the right arm architecture used. 

Finally, what does the compiler output look like, and what’re you using to build?

alejandroImagina, 2018-01-09:
Thanks for the reply.

I've checked the pins, and the PA02 in the Trinket is the PIN 1 in the Arduino IDE (the one I'm using), and the architecture must be the same, because I'm using the SAMD21E17 and the trinket is SAMD21E18, which are the same microcontroller but with different amount of Flash and SRAM (ports and peripherals are the same), so that's why I thought it should be enough to make it worked.

The different is that, if I want to write to the PA02 with the digitalWrite function, I've must call the PIN 2 in the IDE, because the Mattairtech board is defined that way. But since I'm telling the library I'm using the trinket, I defined the PIN 1 in it. (I've also tried defining the PIN 2 and the results were the same)

The compiler output is just an OK, no errors or warnings, and the function is being called correctly, but it simply didn't send anything through the PA02 (or any other pin, because I've checked all of them with a logic analyzer).

So I don't know why it isn't working :(

focalintent, 2018-01-09:
I don’t think the define you’re using is fully succeeding in telling the library what you’re using - can you go into the Arduino settings, turn on verbose compiler output and rebuild - paste the build log into gist.github.com and put the link here.   I suspect there’s another platform definition being set that’s overriding what you’re trying to do.

alejandroImagina, 2018-01-11:
Hi, thanks for the answer.

I've been checking the output in the compiler and you were right, in the code I defined the "Trinket_M0". But when it compiles it send the Arduino_Samd_Zero (or I think so), as it can be seen in the next line:

Compiling library "FastLED"
"C:\Users\Imagina Ingenio\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os  -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD -DF_CPU=48000000L -DARDUINO=10805 -DARDUINO_SAMD_ZERO -DARDUINO_ARCH_SAMD"

The microcontroller in the Arduino Zero is the SAMD21G18, same family and architecture but more memory, ports and Pinout, so maybe that's why I can't get the library to work.

I'll check the PIN definitions for the Arduino_SAMD_Zero in the library, and try to modify them to make it work for my microcontroller.

Thanks for the help.

alejandroImagina, 2018-01-11:
I change the Pinout in the "fastpin_arm_d21.h", and now the Leds works :)

The library now looks like:

`#if defined(ARDUINO_SAMD_ZERO)

#define MAX_PIN 22
_DEFPIN_ARM(  2, 0,  2); _DEFPIN_ARM(  3, 0,  3); _DEFPIN_ARM(  4, 0,  4); _DEFPIN_ARM(  5, 0,  5);
_DEFPIN_ARM(  6, 0,  6); _DEFPIN_ARM(  7, 0,  7); _DEFPIN_ARM(  8, 0,  8); _DEFPIN_ARM(  9, 0,  9);
_DEFPIN_ARM( 10, 0, 10); _DEFPIN_ARM( 11, 0, 11); _DEFPIN_ARM( 14, 0, 14); _DEFPIN_ARM( 15, 0, 15); 
_DEFPIN_ARM( 16, 0, 16); _DEFPIN_ARM( 17, 0, 17); _DEFPIN_ARM( 18, 0, 18); _DEFPIN_ARM( 19, 0, 19); 
_DEFPIN_ARM( 22, 0, 22); _DEFPIN_ARM( 23, 0, 23); _DEFPIN_ARM( 27, 0, 27); _DEFPIN_ARM( 28, 0, 28); 
_DEFPIN_ARM( 30, 0, 30); _DEFPIN_ARM( 31, 0, 31);

#define HAS_HARDWARE_PIN_SUPPORT 1`

And it will works for all the MattairTech boards, because if you have a J or G SAMD21, the only changes will be the MAX_PIN XX and add the new _DEFPIN_ARM( XX, 0, XX)

Thanks to **focalintent** for the help, and I hope that this will be helpful for the ones that want to use this fantastic library with this family boards ;)

closed 2019-01-20 by focalintent

huster-songtao, 2023-11-07:
> I change the Pinout in the "fastpin_arm_d21.h", and now the Leds works :)
> 
> The library now looks like:
> 
> `#if defined(ARDUINO_SAMD_ZERO)
> 
> #define MAX_PIN 22 _DEFPIN_ARM( 2, 0, 2); _DEFPIN_ARM( 3, 0, 3); _DEFPIN_ARM( 4, 0, 4); _DEFPIN_ARM( 5, 0, 5); _DEFPIN_ARM( 6, 0, 6); _DEFPIN_ARM( 7, 0, 7); _DEFPIN_ARM( 8, 0, 8); _DEFPIN_ARM( 9, 0, 9); _DEFPIN_ARM( 10, 0, 10); _DEFPIN_ARM( 11, 0, 11); _DEFPIN_ARM( 14, 0, 14); _DEFPIN_ARM( 15, 0, 15); _DEFPIN_ARM( 16, 0, 16); _DEFPIN_ARM( 17, 0, 17); _DEFPIN_ARM( 18, 0, 18); _DEFPIN_ARM( 19, 0, 19); _DEFPIN_ARM( 22, 0, 22); _DEFPIN_ARM( 23, 0, 23); _DEFPIN_ARM( 27, 0, 27); _DEFPIN_ARM( 28, 0, 28); _DEFPIN_ARM( 30, 0, 30); _DEFPIN_ARM( 31, 0, 31);
> 
> #define HAS_HARDWARE_PIN_SUPPORT 1`
> 
> And it will works for all the MattairTech boards, because if you have a J or G SAMD21, the only changes will be the MAX_PIN XX and add the new _DEFPIN_ARM( XX, 0, XX)
> 
> Thanks to **focalintent** for the help, and I hope that this will be helpful for the ones that want to use this fantastic library with this family boards ;) (or with any other custom board with the PinOut coincident to the PA)

[https://files.seeedstudio.com/wiki/XIAO_WiFi/pin_map-2.png](url)

How to change the Pinout in the "fastpin_arm_d21.h" for [Seeed Studio XIAO ESP32C3](https://wiki.seeedstudio.com/XIAO_ESP32C3_Getting_Started/)?