FastLED/FastLED #549: ESP8266 LED being overridden by FastLED

LeafyDev, 2018-01-14:
Hi, sorry if this is addressed somewhere else, but I have a NodeMCU using WiFi and FastLED to control some Neopixel strips, I make the onboard LED flash (pin 2) during setup phase, but then when I tell it to turn off, it won't ever stay off...

If I remove FastLED from the includes (and therefore all the code that uses it) the LED behaves fine. Yet I've looked through the FastLED code and see that the only place it could be used is the gMaxPowerIndicatorLEDPin... Which I've played with with no success...?

Any ideas? Thanks

focalintent, 2018-01-16:
@kriegsman - know what's up with this?

kriegsman, 2018-01-16:
Huh.  I think I have a NodeMCU around here, I'll see if I can check.  
@LeafyDev: Exactly which NodeMCU board are you using?  I think mine is the older NodeMCU 0.9 revision.  I'd be happy to get a new board and test, but I'd like to know precisely which NodeMCU board you have there.  Thanks!

FaldoAU, 2018-01-16:
Just poking my head in to say: you guys know you're the fucking best,
right?!

LeafyDev, 2018-01-16:
It's a Chinese one I got from Amazon, it exactly looks like a 1.0 and is on Amazon (France) with this name:

Bubudong ESP8266 Serial Port Module Wifi NodeMcu Lua WIFI V3

I'm available to test stuff and screen share with webcam to avoid you buying stuff if needed

kriegsman, 2018-01-16:
Thanks for that info -- I wanted to make sure that I had a board that matched the hardware you have exactly.  Luckily, they're so cheap and so easy to get that I should have my board to test with in just a couple of days.

For reference, this was the image on amazon.fr showing the board... note the rectangular USB/serial converter (CH340) and the LoLin silkscreening.
https://images-na.ssl-images-amazon.com/images/I/61pXFl9MTuL._SL1020_.jpg
and here's the one that I have heading to me for testing
https://images-na.ssl-images-amazon.com/images/I/71XlmUW%2BQDL._SL1200_.jpg
Looks very similar.

In contrast, here are some (newer?) ones with a square USB/serial converter and no LoLin silkscreening:
https://images-na.ssl-images-amazon.com/images/I/71GUeTwFn5L._SL1088_.jpg

So we'll see what actually arrives, but I appreciate this ticket report.  I'll let you know what I find once I have hardware and some time to test it.  Thank you!

LeafyDev, 2018-01-16:
Oh wow, thanks!

I'm adding an image to show you why this bothers me, it's insanely bright :(

[image]

And also the code I use: https://gist.github.com/LeafyDev/e9ff8a4c8ec09ec1604c4fb764626b1a

Let me know if you need anything else :)

kriegsman, 2018-01-16:
Yeah... that's brighter than one might wish.  Thanks for the code and I'll let you know what I find.

marmilicious, 2018-01-17:
Commenting because I'm curious to know what you hopefully discover on this strange one Mark.  

@LeafyDev Yes, those little onboard LEDs are sometimes super bright!  I did use that to my advantage once though, sending it out a little custom light pipe to the outside of a case.
https://www.youtube.com/watch?v=TKM7vcbbOy0&feature=youtu.be&t=21s

pacobyte, 2018-01-19:
@LeafyDev Here's something you can try in the meantime. I use FastLED with a NodeMCU, but mine is what I think is referred to as v2 (it looks more like the last linked image in kriegsman's post above). My board has **two** built-in LEDs—a red one on the dev board and a blue one on the actual ESP8266 (which appears to be the same as the one that's lit in your posted photo).

In order to use the red LED, I simply need to refer to it as `LED_BUILTIN` with no need to `#define` it. So, to turn it on would just be:
``` c
pinMode(LED_BUILTIN, OUTPUT);
digitalWrite(LED_BUILTIN, LOW);
```
I've heard that `LED_BUILTIN` is an auxiliary constant and not a board pin, so if you also have that 2nd LED on your NodeMCU, maybe that one would play nicer with FastLED.

A workaround, I know, but I thought I'd throw that out there.

LeafyDev, 2018-01-19:
I have looked into this yes, but either the LED isn't on my board or
doesn't respond to LED_BUILTIN stuff, I may give it another go this evening
without including any libraries just to see... Thanks for the idea though.

LeafyDev, 2018-01-23:
Hi, quick update, I might be missing something but it may not actually be FastLEDs fault, since this project uses it and I don't get any LED activity at all...

https://github.com/jasoncoon/esp8266-fastled-webserver/

It may be a different library messing with it, but if I comment out FastLEDs includes, it works perfectly as expected.. Interesting

connornishijima, 2018-04-17:
You wouldn't by chance be running FastLED on the GPIO that the built-in LED is tied to, right?

chrisparton1991, 2018-07-05:
You may have solved this by now, but instead of reading the state with `int state = digitalRead(LED_PIN)`, can you try using a global boolean to store the LED state and use that?

Given that the pin is set to output and it has an LED attached, it's possible that reading the pin is always returning LOW.

m2049r, 2020-08-20:
`FastLED.show()` just turns the LED on. Interestingly, if the LED is disabled with `pinMode(LED_BUILTIN, INPUT)` then `FastLED.show()` does nothing!