FastLED/FastLED #552: Problems installing FastLED library

Steviemcc, 2018-01-23:
Hi, 

Sorry if I'm being daft but I'm having trouble installing the library, tried via Arduino Library manager + manually, followed instructions both times but when I try to upload any sketch (even a blank one) it come's up with the following message...

In file included from /Users/McClumpha/Documents/Arduino/FastLED_Tutorial/FastLED_Tutorial_Code/FastLED_Tutorial_Code.ino:2:0:
/Users/McClumpha/Documents/Arduino/libraries/FastLED/FastLED.h:17:21: note: #pragma message: FastLED version 3.001.003
 #    pragma message "FastLED version 3.001.003"
                     ^

The sketch I'm trying to use is an example I found online (chasing LED's, fading etc).

Also when I click include Library all this appears in the window... (FastLED.h is in orange)

#include <bitswap.h>
#include <chipsets.h>
#include <color.h>
#include <colorpalettes.h>
#include <colorutils.h>
#include <controller.h>
#include <cpp_compat.h>
#include <dmx.h>
#include <FastLED.h>
#include <fastled_config.h>
#include <fastled_delay.h>
#include <fastled_progmem.h>
#include <fastpin.h>
#include <fastspi.h>
#include <fastspi_bitbang.h>
#include <fastspi_dma.h>
#include <fastspi_nop.h>
#include <fastspi_ref.h>
#include <fastspi_types.h>
#include <hsv2rgb.h>
#include <led_sysdefs.h>
#include <lib8tion.h>
#include <noise.h>
#include <pixelset.h>
#include <pixeltypes.h>
#include <platforms.h>
#include <power_mgt.h>

I'm a total beginner as you may tell, tried uninstalling etc but the fasted library still shows in the list and does the same thing... 

I'm in a right old pickle. Appreciate anyones help.

Steve.

focalintent, 2018-01-23:
You should only include FastLED.h - not the rest of it - that's including things out of order.

Also - it looks like you have an old version of FastLED installed, the most recent version is something like 3.1.7 or 3.1.8 (this is why I have the library print out the pragma message indicating the library version - it's a bug that the arduino ide shows it in red as an error, but that's on them :)

Steviemcc, 2018-01-24:
Wow, thanks so much for your quick reply, will try now with something basic!

Steviemcc, 2018-01-24:
Still shows the same message once I delete all the other #include(s). Any other thoughts? I'm using a mac by the way if that makes any odds?

focalintent, 2018-01-24:
That message will always be there, the library prints it out while it's being built so that the exact library version shows up in build logs - this way when someone submits a bug report, and includes the build log, i can see immediately what version of the library they're using.

closed 2018-01-24 by focalintent

Steviemcc, 2018-01-24:
Oh I get it... Sorry and thanks very much!