FastLED/FastLED #553: Use of FastLED library in MCP23017 I/O expander pins

vishalgpt579, 2018-01-24:
hello,
This Post is more of a query then an issue. I need to implement some of the FastLED capabilities to My I/O expander Board ( which uses  **MCP23017**,an I/O expander IC). 

But the thing is , the pin access of these ICs are not done like the normal digital pins available on **Arduino**, rather its is done as...
Example:
If there is a single ICs, availiable pins = 16 gpio pins ; that is, each IC has 2 banks of 8 pins. This pins cannot be directly addressed as DATA_ PINS in syntax:
_FastLED.addLeds<NEOPIXEL, DATA_PIN>(leds, NUM_LEDS)_

But,  its mostly done by manually making that pin HIGH and Low example : "10000000" and "00000000" for making the first pin of the bank vary.

So , for using WS2812 at these pins ... the comfort of using FastLED library is removed and then we are left with option of using the basic working mechanism , which is for a single LED : 24 bit of data ... and further even each bit ( '1's and '0's)of that data  should have spec high time and fall time 
as per mentioned in the datasheet https://cdn-shop.adafruit.com/datasheets/WS2812B.pdf ... This is what i believe as my available option.

It would be a real help if you give us some of your opinion regarding this situation. 

**_Are we suppose to make our entire set of programs that will generate the 24 bit rectangular wave by making that pin High and LOW as per mention in WS28123 datasheet ... the similar way FastLED works for WS2812?_**

OR

_**Can there be another easier method way ?**_

P.S:  I've utilize many hours for this problem which has yield no results , Please excuse me for any of my grammatical mistakes

focalintent, 2018-01-25:
You are not going to be able to do this at all with that pin expander - the i2c protocol that this chip uses for communication is not much faster than the ws2812 data rate.  In the time that it takes to send a single byte over i2c to set a single pin (and I bet the protocol for this chip requires more than 1 byte per update) you’be blown through the time requirements for _four_ bits of data that the ws2812 wants.  

What board are you using?  Why not switch to one that has more output pins?

tonton81, 2018-01-25:
what about the 10+ mhz spi variant mcp23s17?
probably still wont be fast, but who knows :)
i2c will never be fast enough for high speed data, thats guarenteed

focalintent, 2018-01-25:
I don’t know that even the 10mhz SPI version would be fast enough - even if it was faster, I’m not sure you could get the fine grained timing control you need.

tonton81, 2018-01-25:
its pretty fast, it also supports constant writing to goio banks without reasserting, that could stream data very fast non stop

focalintent, 2018-01-25:
Again, it’s not just speed, it’s about getting the timing control right.

vishalgpt579, 2018-01-25:
@focalintent , I'm using arduino Mega as a slave controller in prototype stage. Though it has 32  Digital pins , but even that would not be sufficient, since we have more then 60 representative nodes and use of modules other then WS2812. 
_Where each Node  =6 WS2812 LEDs ... which ask for 144 bit of data at 800kHz._

And, I have a question regarding what @tonton81 said about using SPI protocol instead of I2C protocol for faster approach ... as mentioned in MCP23017 datasheet ... I2C protocol can also have a max speed of 1.7MHz , so if its not trouble asking how much smaller the data size should be ....so that even MCP23017 could handle...
Can we decrease the no of LEDs in a single node further (which we are wiling too if that works), can this help ?

focalintent, 2018-01-25:
No, the problem isn’t the number of bits - the problem is the ws2812’s data protocol - which I strongly suspect you won’t be able to drive with the MCP23017 using either i2c or SPI (especially off of just a mega).  

If you switch to using apa102 for data, you won’t have the timing problem for writing data - though you will still have to write your own code for sending the led data out over the MCP.

vishalgpt579, 2018-01-25:
Yes @focalintent,  **apa102**  seems a better and more suitable option in our case ... Thank you @focalintent , your suggestion will surely help us  in our project

closed 2018-01-28 by vishalgpt579

tonton81, 2018-01-28:
dont compare high speed i2c with spi, you’ll be lucky to get 1mhz off wiring, i2c meant for in-ccircuit, however, ive driven SPI off 5 feet cabling at 8MHz, you’ll have to go lower with i2c when cabling, 400 or below, some people go below 100khz...

vishalgpt579, 2018-01-28:
5 feet @tonton81 ?! While observing WS2812 signal's nature, given from Mega we encountered so much distortion( see the image from DSO uploaded below, but it worked smoothly :) )and the length of the wire was just about 10-20 cm. 
[image]

Even our work requires transmission of data signal (now for APA102, thanks to @focalintent ) over the same distance , could you suggests any solution to this noise problem ?
Theoretically, we were taught to increase the S/N ratio ,also a friend suggested to use BJT as Buffer (again , theoretically it will remove distortion )... can any of these would help ?
Also , a last request ...I'm having trouble with my I/O circuit board (made using MCP23017), could you guys  suggest any platform where i can get support ?