FastLED/FastLED #554: Esp32 crashes when connecting internet after FastLED.show() has been used

hutje, 2018-01-25:
All processes that are used with wifi are not usable after one time using FastLED.show(). When I do use a wifi related function the esp32 crashes.
When I don't use FastLED.show() everything works just fine until I use it.

focalintent, 2018-01-25:
@samguyer - thoughts?

samguyer, 2018-01-25:
@hutje Which branch of FastLED are you using?

samguyer, 2018-01-25:
Sounds like a problem with interrupts.

hutje, 2018-01-25:
I am using ws2812. Now somebody told me to put the FASTLED_ALLOW_INTERRUPTS 0 before including the library.
Also without the interrupts, the timing of the leds is bad. Apparently there is another branch of this code where the timing is better for esp32.  https://github.com/samguyer/FastLED

hutje, 2018-01-25:
But the main problem of not able to reconnect to internet is solved... Now I need to find a right timing

hutje, 2018-01-26:
Solved it

mkriegsman, 2018-01-26:
Glad to hear it, @hutje.  Any lessons learned that might be good to share with other folks?

kriegsman, 2018-01-26:
Glad to hear it, @hutje.  Any lessons learned that might be useful to share with other people?

hutje, 2018-01-26:
Yeah, I needed the library of @samguyer. Thanks man! The results 56 inch Digital Clock https://imgur.com/gallery/cJPW2

kriegsman, 2018-01-26:
Looks great!
(Closing the ticket.)

closed 2018-01-26 by kriegsman