FastLED/FastLED #557: How to get ColorPalette to work with P9813 Led Type

markbiasotti, 2018-01-31:
Would like to use the ColorPalette sketch to drive multiple (9ea.) P9813 LED Strip Drivers powering non-programmable LED Tape strips. I have verified that my hardware works with the Blink sketch, but I see that this particular sketch was written for the WS3811 LED Strips and I'm wondering how I can modify the sketch to work with P9813's? My goal is to have my 9ea. led strips cycle and blend thru a predefined set of colors.  Perhaps this sketch was not intended to work for the P9813?  I have tried modifying it for the P9813 in void setup() using the following:

FastLED.addLeds<P9813, DATA_PIN, CLOCK_PIN, RGB>(leds, NUM_LEDS);

but I'm unfamiliar with WS3811 types (assume that LED_PIN is the data pin for the WS3811) and unfamiliar with he "Brightness" command etc.

marmilicious, 2018-02-01:
Hello Mark.  What is the WS3811, do you have a link to a data sheet?
I'm not sure I fully understand what you're wanting to do.  Do you have "dumb" (non-addressable) RGB LED strips that you want to control using P9813 chips?

Also, what "Brightness" command are you referring to?  Can you post your code to gist.github.com and share the link?

markbiasotti, 2018-02-01:
Do you have "dumb" (non-addressable) RGB LED strips that you want to control using P9813 chips?

Yes, that is correct. Sketch from GitHub is attached.

Thanks

Mark

Mark Biasotti
markbiasotti@gmail.com <mailto:markbiasotti@gmail.com>

"Without Me you can do nothing and if you do nothing, it will be without Me.”

Jesus Christ & Dallas Willard

marmilicious, 2018-02-01:
Hmm, I don't see any code or link to your code.

If you have this, as you noted above, in your code's setup() section:
FastLED.addLeds<P9813, DATA_PIN, CLOCK_PIN, RGB>(leds, NUM_LEDS);
then that should be sending the proper signal for the P9813 chip.

I guess you're running the RGB outputs from the P9813 to some MOSFETS for controlling the R,G,B channels on the basic LED strip?

markbiasotti, 2018-02-01:
Hi Mark,

I found this page <https://github.com/FastLED/FastLED/wiki/Controlling-leds> and I’m studying it and hopefully understand the FastLED commands and write my own sketch. To answer your question, yes, I have 9ea. P9813 LED Strip MOSFET driver boards that I’m driving 9 non-programable RGB LED strips. I would like to see if I can get the ColorPalette sketch to work with them (driver boards) but the sketch is written for the WS281type of led controller.  the ColorPallete arduino sketch is attached.

thanks

Mark

Mark Biasotti
markbiasotti@gmail.com <mailto:markbiasotti@gmail.com>

"Without Me you can do nothing and if you do nothing, it will be without Me.”

Jesus Christ & Dallas Willard

> Hmm, I don't see any code or link to your code.
>
> If you have this, as you noted above, in your code's setup() section:
> FastLED.addLeds<P9813, DATA_PIN, CLOCK_PIN, RGB>(leds, NUM_LEDS);
> then that [...]

#include <FastLED.h>

#define LED_PIN     5
#define NUM_LEDS    50
#define BRIGHTNESS  64
#define LED_TYPE    WS2811
#define COLOR_ORDER GRB
CRGB leds[NUM_LEDS];

#define UPDATES_PER_SECOND 100

// This example shows several ways to set up and use 'palettes' of colors
// with FastLED.
//
// These compact palettes provide an easy way to re-colorize your
// animation on the fly, quickly, easily, and with low overhead.
//
// USING palettes is MUCH simpler in practice than in theory, so first just
// run this sketch, and watch the pretty lights as you then read through

// the code.  Although this sketch has eight (or more) different color schemes,
// the entire sketch compiles down to about 6.5K on AVR.
//
// FastLED provides a few pre-configured color palettes, and makes it
// extremely easy to make up your own color schemes with palettes.
//
// Some notes on the more abstract 'theory and practice' of
// FastLED compact palettes are at the bottom of this file.

CRGBPalette16 currentPalette;
TBlendType    currentBlending;

extern CRGBPalette16 myRedWhiteBluePalette;
extern const TProgmemPalette16 myRedWhiteBluePalette_p PROGMEM;

void setup() {
    delay( 3000 ); // power-up safety delay
    FastLED.addLeds<LED_TYPE, LED_PIN, COLOR_ORDER>(leds, NUM_LEDS).setCorrection( TypicalLEDStrip );
    FastLED.setBrightness(  BRIGHTNESS );
    
    currentPalette = RainbowColors_p;
    currentBlending = LINEARBLEND;
}

void loop()
{
    ChangePalettePeriodically();
    
    static uint8_t startIndex = 0;
[... 113 lines cut ...]
// Normally, in computer graphics, the palette (or "color lookup table")

// has 256 entries, each containing a specific 24-bit RGB color.  You can then
// index into the color palette using a simple 8-bit (one byte) value.

// A 256-entry color palette takes up 768 bytes of RAM, which on Arduino

// is quite possibly "too many" bytes.
//
// FastLED does offer traditional 256-element palettes, for setups that

// can afford the 768-byte cost in RAM.
//
// However, FastLED also offers a compact alternative.  FastLED offers

// palettes that store 16 distinct entries, but can be accessed AS IF
// they actually have 256 entries; this is accomplished by interpolating

// between the 16 explicit entries to create fifteen intermediate palette
// entries between each pair.
//
// So for example, if you set the first two explicit entries of a compact 
// palette to Green (0,255,0) and Blue (0,0,255), and then retrieved 
// the first sixteen entries from the virtual palette (of 256), you'd get
// Green, followed by a smooth gradient from green-to-blue, and then Blue.

kriegsman, 2018-02-01:
I've used those driver boards before, and they work fine with FastLED.  
(e.g http://www.dx.com/p/full-color-rgb-led-strip-driver-module-for-arduino-blue-black-314667 )

Each driver board is basically "one LED" as far as the library is concerned.  So if you have nine of these driver boards, you'll want to define NUM_LEDS as 9, and use leds[0] .. leds[9] for setting the colors.

We like to use this github ticketing system just for bug reports in the library itself.  For general help questions like this, use the FastLED discussion group here http://fastled.io/+   Thanks!

kriegsman, 2018-02-01:
Since there's no library bug report here, I'm going to close this ticket.  The "how to" conversation can continue on the G+ group http://fastled.io/+ ... and I suspect you'll get up and running quickly there!

closed 2018-02-01 by kriegsman

markbiasotti, 2018-02-01:
Thanks Mark for the reply.  Yes, what you say about how to use the P9813 as one led is the same conclusion that I drew and I will give this a try. Sorry for posting in the wrong section.

thanks

Mark

Mark Biasotti
markbiasotti@gmail.com <mailto:markbiasotti@gmail.com>

"Without Me you can do nothing and if you do nothing, it will be without Me.”

Jesus Christ & Dallas Willard

kriegsman, 2018-02-01:
No problem, hope you get it up and running quickly!