FastLED/FastLED #559: Discord time of 2 or more uno boards after a while

hamsafar, 2018-02-01:
Hi,
I have 2 uno board and program them to some effects. After run them at a same time,after a while (about 30 secends) missed coordination!
what is my problem from?
Thanks.

Regards
AliReza

focalintent, 2018-02-01:
Two things compounding - one is that the clocks on these devices aren’t terribly accurate and are prone to drifting.  The other is that they use interrupts to drive the clock, and interrupts are disabled when writing out ws2812 style leds - FastLED attempts to adjust the time accordingly to compensate, but it’s very approximate (still better than losing a few dozen milliseconds every tome show is called).  

If you need time sync between two devices you will either need to use Real Time Clocks attached to each to keep time accurate, or use some mechanism to sync (eg have the unos delay calling FastLED.show until each of them have raised a pin high)

hamsafar, 2018-02-02:
Thanks.
I used external crystal 16.000000 Mhz for them! I enabled interrupt(#define FASTLED_ALLOW_INTERRUPTS 1) but problem is too.
For sync device I can't connect device to to each other because they have distance to gether!
For sync them I used form Ac(50Hz) power and turned it to square pulse 5 volt and connect it to T1(D5).
And calc animations time and set counter to desire time and when timer overflow occur, return to begin of animations.
How is this Idea?
But my problem is that my animations time is about 10 minuits, and during of this 10 miniuts micros coordinate missed!

focalintent, 2018-02-02:
Interrupts are only allowed on arm and esp platforms, the avr is too low powered, and your crystals may still have drift between each other and there’s still the issue of the timer interrupts being disabled (which you could mitigate by using something like apa102 instead).  

You should look into using an external RTC (real time clock) on both devices - these will keep accurate time and you can use them to determine the time for syncing animations.

hamsafar, 2018-02-03:
I will test external RTC.
Thanks

closed 2018-02-03 by hamsafar