FastLED/FastLED #560: RE: Led color changes after 50 leds though the hex passed is same

siddharthtaunk, 2018-02-04:
Hi,

I am working with FastLED and I face the following issue

I have an array declared as following
**`CRGB color_array[] = {CRGB::Green, CRGB::Red, CRGB::Blue, CRGB::DeepPink, CRGB::Yellow, CRGB::LavenderBlush, CRGB::White, CRGB::DeepSkyBlue, CRGB::Red, CRGB::Moccasin, CRGB::White, CRGB::LemonChiffon};`**

I have declared
`#define NUM_LEDS 500`

but It does not work after 50 lEDs the color starts changing and flickering.
I am using 5V 2A supply to my WS2812 Strips

Kindly help with this issue.
Its a bit urgent

Thanks and regards.
Siddharth

focalintent, 2018-02-04:
I'd need to see the rest of your code to see if there's an issue there  - you also don't mention anything about what hardware you are using - or how it is wired up, or how you are managing or wiring power.  2A is very likely not enough to drive 500 leds, however - where are you injecting power into the strips?   It is possible you're running into power/voltage issues.

siddharthtaunk, 2018-02-04:
Hi,

Thanks for the reply, I am using Arduino Uno (atmega328)(for 50 led) and Arduino Mega(amtge 2560)(for 500 LEDs)

Here is the brief excerpt from the code

```
#include "FastLED.h"

#define NUM_LEDS 500    //change to 64 to test
#define DATA_PIN 2

#define LED_TYPE    WS2811
#define COLOR_ORDER GRB
#define FRAMES_PER_SECOND  120

CRGB leds[NUM_LEDS];

CRGB color_array[] = {CRGB::Green, CRGB::Red, CRGB::Blue, CRGB::DeepPink, CRGB::Yellow, CRGB::LavenderBlush, CRGB::White, CRGB::DeepSkyBlue, CRGB::Red, CRGB::Moccasin, CRGB::White, CRGB::LemonChiffon};
void setup() {

  Serial.begin(9600);
  // put your setup code here, to run once:
  delay(3000);
    FastLED.addLeds<LED_TYPE,DATA_PIN,COLOR_ORDER>(leds, NUM_LEDS).setCorrection(TypicalLEDStrip);
    FastLED.setBrightness(brightness);
}
void setup() {

  Serial.begin(9600);
  // put your setup code here, to run once:
  delay(3000);
    FastLED.addLeds<LED_TYPE,DATA_PIN,COLOR_ORDER>(leds, NUM_LEDS).setCorrection(TypicalLEDStrip);
    FastLED.setBrightness(brightness);
}
void setup() {

  Serial.begin(9600);
  // put your setup code here, to run once:
  delay(3000);
    FastLED.addLeds<LED_TYPE,DATA_PIN,COLOR_ORDER>(leds, NUM_LEDS).setCorrection(TypicalLEDStrip);
    FastLED.setBrightness(brightness);
}

void pattern15()
{
    smoothflash(12,SMOOTH_FLASH_SPEED);
      stat = '1';
}

void smoothflash(int count, int speed){
 for(int i=0; i<count; i++){
      allColor(color_array[i]);
    delay(speed);
  }
}

void allColor(CRGB c){
  for(int i=0; i<NUM_LEDS; i++)
      leds[i] = c;
  FastLED.show();
}
```
connections
**VCC** from 5V 2A supply to Arduino mega and the WS2812 strip
**DATA** only to the strip fron pin 2 of Arduini
**GND** from 5V 2A supply to Arduino mega and the WS2812 strip

Kindly help

focalintent, 2018-02-04:
You need to add power to WS2812 leds every 50-100 leds, because of voltage drop along the strips.  Also, 500 WS2812 leds are going to use 0.5A when completely off - and draw as much as 20amp at full white - and 6+ amps at full Red, Green, or Blue - so your power supply is way too small for 500 leds, at least without setting the brightness to something really low.  However, again, you also have a different potential issue with voltage drop - so you should be adding power every 50-100 leds, give or take.

siddharthtaunk, 2018-02-04:
So if I have to calculate,

20mA per LED 
so its 20mA * 150 or 20mA * 500
that comes to 3000mA or 3A+0.5A(for the drivers) for 150 leds
and 10000mA or 10 A(+1A for drivers) for 500 leds

Please correct me if I am wrong.

Thanks for the prompt replies.
Siddharth

focalintent, 2018-02-04:
20mA if you only have one of Red, Green, or Blue at full brightness - at full white it is more like 60mA - and the chip draws 1-2mA on its own when none of the leds are on.

siddharthtaunk, 2018-02-04:
Apart from the above there is one more issue the Arduino mega hangs and reboots adter a patters which was nkot happening with atmega328(UNO) I have used the **stk500boot_v2_mega2560** bootloader
Fuse bits : Didn't change

**This may be due to less current to the mega to reboot?**(Suspecting)

regards.
Siddharth

focalintent, 2018-02-04:
That could certainly happen.  Another cause could be a bug in your code that has you overwritten random memory...

siddharthtaunk, 2018-02-04:
and regarding the color change? that may be due to the lack of current?

focalintent, 2018-02-04:
Yeah - too much voltage drop can mess with the ability of the leds to read the signal properly

siddharthtaunk, 2018-02-06:
Hi,

I still have issues with the color changing, not for R, G and B, but for the other colors they fade to a different colors after 75 LEDs.

Now my supply is an hefty 5V 20A

Is there a problem with the colr
`CRGB color_array[] = {CRGB::Green, CRGB::Red, CRGB::Blue, CRGB::DeepPink, CRGB::Yellow, CRGB::LavenderBlush, CRGB::White, CRGB::DeepSkyBlue, CRGB::Red, CRGB::Moccasin, CRGB::White, CRGB::LemonChiffon};
`

Kindly help

yesyesuk, 2018-02-06:
Have you tried to provide power (5V) in a few places along the strip. The thin, narrow trace for power supply on these strips is not enough to carry that much current and voltage will drop along the strip. At a certain distance it will not be enough voltage for the LEDs to work correctly.
If you have problems from LED number 75, try adding power after 60 LEDs or so.

siddharthtaunk, 2018-02-07:
Yes the above solution works great. 

How do I get it up and running on Arduino Mega
I have the following bootloader in Arduino mega - **stk500boot_v2_mega2560**
and I have the array size of 500 as seen in the post above `#define NUM_LEDS 500`  the code works hay way and there is no color sequence being followed, its all random.

I strongly guess I don't have the right bootloader.

Kindly help
Siddharth

closed 2018-06-23 by focalintent