FastLED/FastLED #561: 3.1.8 Release not available in Arduino Library Manager

maddogjt, 2018-02-05:
The Arduino library manager is still only showing the 3.1.6 release, although it looks like work was done to publish a 3.1.8 release. 

I'm not familiar with the library release process, but according to the docs that I found [here](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-publish-a-new-release-once-my-library-is-in-the-list), [library.properties](https://github.com/FastLED/FastLED/blob/f6c4e51ac7ce7ef7813eda91bb70dca5a3827634/library.properties#L2) may be in need of an update.

kriegsman, 2018-02-05:
Thank you, I’ll get on that ASAP.

per1234, 2018-02-08:
The library.properties `version` value must be updated before the tag is created. You would need to either redo the 3.1.8 tag or release a new version. 

The same issue is also the cause of 3.1.7 not being available in Library Manager:
https://github.com/FastLED/FastLED/blob/3.1.7/library.properties#L2

kriegsman, 2018-02-08:
Yep. I’m going to tag 3.1.9. 
And possibly add a checklist file to the library for future releasers so we make sure to update everything all at once. Sorry for the delays; we’ll get it straightened out quickly now.

closed 2019-01-20 by focalintent