FastLED/FastLED #565: Request for sawtooth wave to be added to FastLED. You already have a triangle wave.

atuline, 2018-02-10:
Please see the following diagram:

http://cds.linear.com/image/WaveformShapes.png

As a matter of fact, a sawtooth wave in both directions would even be better.

Thanks,

Andrew

EmerickPoulin, 2018-02-13:
[image]

credit to the user who posted this image at this site:https://www.muffwiggler.com/forum/viewtopic.php?p=1144128

atuline, 2018-02-14:
Thanks. I'll study and play around with those more carefully.

R-OLE, 2018-09-18:
Also please remove the "aka sawtooth" phrase from the wiki entry on [wave functions](https://github.com/FastLED/FastLED/wiki/FastLED-Wave-Functions) to:
1. Not lead folks astray that are just learning these concepts
2. Avoid confusion with the aptly named functions listed in this issue