FastLED/FastLED #567: Faster "moving pixel" on WS2812 matrix; FastLED.show() is slow -.-"

slowpoek, 2018-02-13:
Hello everyone, i would highly appreciate any meaningful thought regarding a timing problem with my WS2812-matrix. It's not a precision problem as in "is my ``delay(1000)`` really 1000ms long?", it's about pure speed :)

**project outline / motivation:**
I intend to measure the delay of multiple cameras triggered by the same release signal. I expect 0-10 ms delay values. Therefore, I thought of a quick LED dot toggling through a 256-pixel WS2812 matrix as fast it can. I capture the matrix with my cameras and from the resulting images I can deduct the camera delay (Picture1: "The dot is at line8, pos3.", Picture2: "It clearly is at line8, pos5, screw you!")

**problem / symptoms:**
Even in a minimal case, I can't beat a time delay of 9ms from frame-to-frame (LEDposX-to-LEDposX+1) on my moving-dot-matrix. Correct me if I'm wrong, in my understanding that corresponds to a frame rate of 111Hz, while the technical limit of the WS2812 itself should be at 400Hz. 

**minimal:**
```#include "FastLED.h"
#define PIXEL_TYPE NEOPIXEL
#define NUM_LEDS 256
#define DATA_PIN 6

CRGB leds[NUM_LEDS];

void setup() { FastLED.addLeds<PIXEL_TYPE, DATA_PIN>(leds, NUM_LEDS); }

void loop() {
        for(int dot = 0; dot < NUM_LEDS; dot++) { 
            leds[dot] = CRGB::Blue;
            FastLED.show();
            FastLED.clear();
        }
 } 
```

**sloppy eigendiagnosis:**
So obviously, the time-consuming part is NOT preparation of pixel (see video in link below, a lot more to do, yet same speed), but the actual sending through the FastLED.show() command. Adding a ``delay(1)`` does not change anything, a ``FastLED.delay(1)`` makes it even slower (18ms / 56Hz). Right now, I'm running the code with no delay at all. Played around with "EVERY_N_"-stuff, Adafruit library (boo!), didn't help.

**setup:**
Hardware: Arduino UNO, 5V/16A-PWRsupply, WS2812-Matrix (256 LEDs), 1000uF capacitor, 470Ohm resistor
Software: FastLED 3.1.6
Video snippet of code I actually intend to use: [https://streamable.com/vtbze](https://streamable.com/vtbze)
Measurement: stopwatch, 10 cycles => divide by 10*256 => "frame time" of roughly 9ms 

**QUESTION 1**
Basically, why is it that slow? Or in other words, is there a way to update LEDs faster? In my opinion, each new pixel movement requires a FastLED.show() call because the position is updated and it is a different set of information than before. Can you "magically speed up" that function call? Not an experienced person here, obviously...

**QUESTION 2**
Does anyone have a smart thought on how to create another ON/OFF-pattern or some other sort of visually distinguishable pattern that runs quicker than 9ms/frame and that maybe doesn't require a FastLED.show() call every iteration?

Thankful for any advice and/or opinion! If you need any more information, drop a note, I'll post everything that may help the case.

yesyesuk, 2018-02-13:
Hi,
have you tried with fewer LEDs? The more LEDs the longer it takes to push the data out to the LEDs. Try with 100 LEDs and see what results you get. You don't need to cut the strip, just set NUM_LEDS to 100 and the other 156 LEDs will just stay off.

slowpoek, 2018-02-13:
That is not only a pretty intuitive explanation, and not only does it also deliver a nice-and-easy quickfix strategy, your answer itself turned out to be a really quick one itself! Thank you very much!!

I found the following: 
```
NUM_LEDS 256    // frame time     9 ms
NUM_LEDS  96    // frame time   4.3 ms
NUM_LEDS  64    // frame time   3.4 ms
NUM_LEDS  48    // frame time   3.1 ms
```
So depending on the time resolution I intend to measure I can adjust my matrix and give results as in "Those two cameras see the same pixel, so their release had no delay within an uncertainty of 3ms."
Keeping the physical limit of the Neopixels ("less than 2.5ms may hurt your cells") in mind, that is all I want! Case closed for now.

Again, thank you!

gnkarn, 2018-02-13:
I think you should review the code , fast led show and fast led clear should not be inmediately one after the other , a better solution would be just to clear the previous led so you will always have only one pixel on at any time .

slowpoek, 2018-02-15:
Good reminder! 
Since I'm happy with yesyesuk's quickfix strategy, I just wanted to share one version of my final code, in case someone has a use for it. Frame times are determined by visually observing multiple cycles and calculating things down (translates to: "there might be a more precise way..."), so better re-evaluate timing on your own setup. Here's a "frame time 10ms" version, the edges of any 16x...-matrix are highlighted.
```
// LED-Skript to visually distinguish a time delay in increments of 10 ms 
// 
// lights up 256 cells 
// which is equal to 16 row a 16 cells
// a delay(2) is added to match the 10ms frame time best
// code is flexible for any 16x...-matrix, needs some changes for other setups

#include "FastLED.h"
#define COLOR_ORDER RGB
#define PIXEL_TYPE NEOPIXEL
#define DATA_PIN 6
#define NUM_LEDS 256
#define NUM_LEDS_PART 256

#define DELAY 2                             // "2" to be used for 10ms frame time

uint8_t hue = 0;
uint8_t gHueDelta = 192;

CRGB leds[NUM_LEDS_PART];
CRGB leading_colour( 64,0,64 );             //PURPLE

int corner1 = 0;
int corner2 = 15;
int corner3 = NUM_LEDS_PART - 16;
int corner4 = NUM_LEDS_PART - 1;

void setup() { 
  FastLED.addLeds<PIXEL_TYPE, DATA_PIN>(leds, NUM_LEDS_PART); 
}

void loop() {
   
  //ON-Loop
    for(int dot=0; dot<NUM_LEDS_PART; dot++){
              
        leds[dot]=CRGB( leading_colour );   //Leading LED
        
        leds[corner1]=CRGB( 32,32,32 );
        leds[corner2]=CRGB( 32,32,32 );
        leds[corner3]=CRGB( 32,32,32 );
        leds[corner4]=CRGB( 32,32,32 );
        FastLED.show();
        delay(DELAY);
        leds[dot]=CHSV(hue, 255, 16);
        }
  
  //OFF-Loop  
  for(int dot=0; dot<NUM_LEDS_PART; dot++){
    
        leds[dot]=CRGB( leading_colour );   //Leading LED
        
        leds[corner1]=CRGB( 32,32,32 );
        leds[corner2]=CRGB( 32,32,32 );
        leds[corner3]=CRGB( 32,32,32 );
        leds[corner4]=CRGB( 32,32,32 );
    FastLED.show();
    delay(DELAY);
    leds[dot]=CRGB(0,0,0);
  }
  hue = hue + gHueDelta;
}
```

Thanks again!

closed 2018-02-15 by slowpoek

tip32a, 2019-02-09:
I would like to re open this question.  I see that the original poster found an answer that helped him.  But it does not answer the posted question.  I need to use all 144 leds and want them the turn on 1 at a time.  1 out of 144.  straight up the strip as fast as possible.  To create a lightning effect.   
I am using an Arduino Nano

focalintent, 2019-02-09:
The fastest code for this is going to be something like:

```
#include <FastLED.h>
#define NUM_LEDS 144
#define PIN 6
CRGB leds[NUM_LEDS];

void setup() { FastLED.addLeds<WS2812,PIN>(leds, NUM_LEDS); }
void show() { 
    static byte pos = 0;
    if(pos == NUM_LEDS) { 
      // we've lit all the less, now turn them all off and start again
      for(int i = 0; i < NUM_LEDS; i++ ) { leds[i] = CRGB::Black; }
      pos = 0; 
   }
    leds[pos++] = CRGB::White;
    FastLED.show();
}
```

Now - here's where math comes into play.  It takes 30µs to write the led data for a single led (1.25µs per bit, 24 bits per pixel) - which means it will take 4.320ms to write a single frame's worth of data - which means it will take 622ms to fill the strip.

Whether or not that's fast enough for you is up to you :)

tip32a, 2019-02-10:
Thanks for your reply,

Where did all the leds get lit?
There is only 1 FastLED.show();  immediately after the 
leds[pos++] = CRGB::White;  This code is confusing if it is correct.

focalintent, 2019-02-10:
In this line:

```
leds[pos++] = CRGB::White;
```

The pos++ increments the value of pos by one.  

While in This line:

```
static byte pos = 0;
```

The ‘static’ means that this variable keeps its value when the function loop ends.  And on the Arduino loop is called repeatedly.  

So the first time loop is called, pos is equal to zero and the first item in the led array will be set to white.  The pos will be incremented and have a value of 1 - then the next time loop is called pos will have a value of 1 and so now the second led will be set to white, and so on and so on.  

Until the value of pos is the same as NUM_LEDS (144 in your case) - at which point all the leds are reset to black and pos is set to 0 and it starts all over again.

yesyesuk, 2019-02-10:
`void show() {`

should probably be

`void loop() {`

focalintent, 2019-02-10:
Whoops - sorry about that!  (This is what I get for typing out code on a phone on a bus :)

tip32a, 2019-02-11:
so each time the routine loops the led array is recreated so by default all are black.  you are simply setting 1 led to white.  then displaying.  then incrementing the next led to light.  the loop restarts clearing the array and setting the next to white.  If I am reading this correctly?

yesyesuk, 2019-02-11:
The way I read it, it fills the strip white 1 LED at a time. So the white gets 1 LED longer with each loop. When the length of the strip has been reached, all LEDs are set to black and then it starts again.

dimitre, 2019-02-15:
@focalintent thanks for the µs information. how about the new frame information? is it only a few bytes in length or the same size as the packet itself? Thanks

tobi01001, 2019-02-19:
As far as I know (please correct me if I am wrong):
Any new frame (or any change to a led) requires the complete data to be written out to the strip.
Once you defined your (global) LEDs array (moreover the fastled controller class constructor) the size of the new frame information is fixed to the size of the leds array...