FastLED/FastLED #569: Error while tying to compile

BANOnotIT, 2018-02-16:
I'm on ubuntu 16.04.3. I wanna compile code with FastLED, but it throws error:

```
Arduino: 1.8.5 (Linux), Плата:"Arduino/Genuino Uno"

In file included from sketch/FastLED/platforms/avr/fastled_avr.h:6:0,
                 from sketch/FastLED/platforms.h:32,
                 from sketch/FastLED/FastLED.h:55,
                 from /home/ <***> /main/main.ino:12:
sketch/FastLED/platforms/avr/clockless_trinket.h:4:24: fatal error: controller.h: No such file or directory
 #include "controller.h"
                        ^
compilation terminated.
exit status 1
```
Compiler command:
```
"/home/***/ard/arduino-1.8.5/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/home/***/ard/arduino-1.8.5/hardware/arduino/avr/cores/arduino" "-I/home/***/ard/arduino-1.8.5/hardware/arduino/avr/variants/standard" "-I/home/***/ard/arduino-1.8.5/hardware/arduino/avr/libraries/SoftwareSerial/src" "/tmp/arduino_build_257879/sketch/main.ino.cpp" -o "/tmp/arduino_build_257879/preproc/ctags_target_for_gcc_minus_e.cpp"
```
I don't know how to fix it... Can anybody help me?

BANOnotIT, 2018-02-16:
I tried 2 versions: this error is throwing on 3.1.6 and the same error throwed on 3.1.8

focalintent, 2018-02-16:
FastLED should be installed in your arduino’s libraries directory, not with your sketch.

Skuschy, 2018-02-24:
I have a very similar problem I think.
I get the following error:
`Documents\Arduino\Sound_STRIP\Sound_STRIP.ino:27:19: fatal error: FastLED: No such file or directory

 #include <FastLED>

                   ^

compilation terminated.

exit status 1
`
I have searched many different websites, but I haven't found anyone yet having this errormessage. How do I fix this?

focalintent, 2018-02-24:
It's FastLED.h - not FastLED - and that's not this issue.

closed 2018-02-24 by focalintent